isaac-sim/IsaacLab #333: [Bug Report] Same container names

Divelix, 2024-03-28:
When I try to run `./docker/container.sh start` it throws:
```
"services.orbit-ros2": container name "orbit" is already in use by "services.orbit-base": invalid compose project
```

It appears that both services (base and ros2) use the same container name `orbit`. If I change `container_name` in `orbit-ros2` service to smth other than `orbit` (I used `orbit-ros2` name) it works fine.

jsmith-bdai, 2024-03-28:
> When I try to run `./docker/container.sh start` it throws:
> ```
> "services.orbit-ros2": container name "orbit" is already in use by "services.orbit-base": invalid compose project
> ```
>
> It appears that [...]

Hello, thanks for posting!

We are aware of this issue (it's due to different versions of docker-compose supporting different functionality)and have a fix that should be merged this week. Your fix is close to what we are doing on our side, too to accommodate different docker compose versioning. Sorry for any inconvenience!

Mayankm96, 2024-04-12:
@pascal-roth @hhansen-bdai Is this good to close now?

Mayankm96, 2024-04-26:
I believe this is resolved now. Please re-open the issue if the problem persists.

closed 2024-04-26 by Mayankm96