isaac-sim/IsaacLab #334: [Bug Report] Container install failed

swstbecrpgmail, 2024-03-31:
I download the source code of orbit from github, and I run "./docker/container.sh start". 
It reports that "ERROR [orbit-base base 7/9] RUN --mount=type=cache,target=/root/.cache/pip     /workspace/orbit/orbit.sh --install --extra".
The errors are:
 "ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
nvidia-srl-usd 0.13.0 requires tqdm<5.0.0,>=4.63.0, which is not installed.
16.17 nvidia-srl-usd 0.13.0 requires usd-core<24.00,>=21.11, which is not installed.
16.17 nvidia-srl-usd-to-urdf 0.5.0 requires usd-core<24.00,>=21.11, which is not installed.
16.17 aiobotocore 1.2.0 requires botocore<1.19.53,>=1.19.52, but you have botocore 1.20.49 which is incompatible."

"ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
30.19 nvidia-srl-usd 0.13.0 requires usd-core<24.00,>=21.11, which is not installed.
30.19 nvidia-srl-usd-to-urdf 0.5.0 requires usd-core<24.00,>=21.11, which is not installed.
30.19 WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
"

“error: 7990 bytes of body are still expected
71.88   fetch-pack: unexpected disconnect while reading sideband packet
71.88   fatal: early EOF
71.88   fatal: fetch-pack: invalid index-pack output”

“ERROR: Could not find a version that satisfies the requirement rsl-rl (unavailable) (from omni-isaac-orbit-tasks[all]) (from versions: none)
79.82 ERROR: No matching distribution found for rsl-rl (unavailable)
79.83 WARNING: You are using pip version 21.2.1+nv1; however, version 24.0 is available.
79.83 You should consider upgrading via the '/isaac-sim/kit/python/bin/python3 -m pip install --upgrade pip' command.
79.95 There was an error running python

failed to solve: process "/bin/bash -c ${ORBIT_PATH}/orbit.sh --install --extra" did not complete successfully: exit code: 1”

[image]
[image]
[image]
[image]

swstbecrpgmail, 2024-03-31:
When I download and install the old version of orbit docker and change the container name "orbit-ros", it works well. But when I try to install the latest version of orbit docker, it failed.

Mayankm96, 2024-04-04:
Please check the latest main. If the issue persists, feel free to re-open the issue. Thank you!

commit 8a9c362c (2023-12-21): Fixes installation guide script path (#335)

After the great tutorial refactor of 2023, the installation guide wasn't
updated. This PR updates the path there.

Please include a summary of the change and which issue is fixed. Please
also include relevant motivation and context.
List any dependencies that are required for this change.

Fixes #334 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./orbit.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My [... cut]
closed 2024-04-04 by Mayankm96