isaac-sim/IsaacLab #335: [Bug Report] Can‘t load quadruped omni.usd correctly

swstbecrpgmail, 2024-04-01:
When I run "./orbit.sh -p source/standalone/demos/quadrupeds.py --livestream 1", it seem that it can't load usd correctly and reports value error as follows.
 
[image]
 
But when I run "./orbit.sh -p source/standalone/demos/quadrupeds.py --headless", the livestream client is able to display the scene.
[image]

swstbecrpgmail, 2024-04-01:
If these two lines in app.py are commended out, then the robots can be loaded correctly.
[image]

pascal-roth, 2024-04-08:
thanks for noticing this issue, @hhansen-bdai you looked most into the livestreaming options, any idea?