isaac-sim/IsaacLab #337: Command and Action

Nimingez, 2024-04-01:
Thank you for you great work！
Robot reach task， reach_env_cfg 
CommadCfg UniformPoseCommand and ActionCfg
I noticed that your command provides the end-effector position for the robotic arm and outputs the action. Where does the process of inverse kinematics take place?

zoctipus, 2024-04-08:
You can take a look at 
```'source/extensions/omni.isaac.orbit/omni/isaac/orbit/controllers/differential_ik.py'```
method:
```_compute_delta_joint_pos```

The implementation how ik is calculated is there

Mayankm96, 2024-04-19:
Exactly. The action term we are using there is [DifferentialInverseKinematicsActionCfg](https://isaac-orbit.github.io/orbit/source/api/orbit/omni.isaac.orbit.envs.mdp.html#omni.isaac.orbit.envs.mdp.actions.actions_cfg.DifferentialInverseKinematicsActionCfg). It converts pose absolute or relative commands to joint commands and applies them into the simulation.

closed 2024-07-02 by Mayankm96