isaac-sim/IsaacLab #338: [Bug Report] Docker run example fail

ruoshiwen, 2024-04-01:
### Describe the bug

I installed the container following the instruction and then run the example `log_time.py` but failed.

### Steps to reproduce

First, I run 
`./docker/container.sh start`

then a detached docker is started. I use vscode docker plugin to attach a vscode to that container and navigate to the /workspace/orbit directory, and then run

`./orbit.sh -p source/standalone/tutorials/00_sim/create_empty.py --headless`

```
Then I get the following error:

`Traceback (most recent call last):
  File "/workspace/orbit/source/standalone/tutorials/00_sim/create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/workspace/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app/app_launcher.py", line 110, in __init__
    self._create_app()
  File "/workspace/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app/app_launcher.py", line 491, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 237, in __init__
    app_version_core, _, _, _, _, _, _, _ = get_version()
  File "/isaac-sim/exts/omni.isaac.version/omni/isaac/version/version.py", line 63, in get_version
    with open(f"{app_start_folder}/VERSION", encoding="UTF-8") as f:
FileNotFoundError: [Errno 2] No such file or directory: '//VERSION'`
```

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 3acff]
- Isaac Sim Version: [2023.1.1-rc.8+2023.1.688.573e0291.tcroot]
- OS: [e.g. Ubuntu 22.04]
- GPU: [e.g. RTX 3070Ti]
- CUDA: [e.g. 12.0]
- GPU Driver: [525.85.05]

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

ruoshiwen, 2024-04-01:
I checkout to the commit [0865cf6b2981e6], running example succeeded.

jsmith-bdai, 2024-04-01:
Thank you for posting - we are working on fixing this

hhansen-bdai, 2024-04-02:
Hello, this is fixed by commit `dfa787`. Thank you for your issue!

closed 2024-04-04 by Mayankm96