isaac-sim/IsaacLab #339: [Bug Report] AppLauncher get_version is pathing to wrong place "../.local/share/ov/pkg/VERSION"

Ltesfaye, 2024-04-01:
### Describe the bug

With the updated simulation_app launcher version none of the existing tutorial or personal project simulation apps launch. This leads to the following error: `FileNotFoundError: [Errno 2] No such file or directory: '/home/pcgadmin/.local/share/ov/pkg/VERSION'`

### Steps to reproduce

1. Download a clean instance of orbit repo and install 
2. run any of the tutorial files using `./orbit.sh -p source/standalone/demos/bipeds.py `

In my case I've run the following:

- All the tutorials in 01 and standalone demo

```Traceback (most recent call last):
  File "/home/pcgadmin/Documents/pull_request/orbit/source/standalone/demos/bipeds.py", line 28, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/home/pcgadmin/Desktop/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app/app_launcher.py", line 110, in __init__
    self._create_app()
  File "/home/pcgadmin/Desktop/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app/app_launcher.py", line 491, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/pcgadmin/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 237, in __init__
    app_version_core, _, _, _, _, _, _, _ = get_version()
  File "/home/pcgadmin/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.version/omni/isaac/version/version.py", line 63, in get_version
    with open(f"{app_start_folder}/VERSION", encoding="UTF-8") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pcgadmin/.local/share/ov/pkg/VERSION'
2024-04-01 17:01:33 [11,406ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/pcgadmin/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-04-01 17:01:33 [11,410ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-04-01 17:01:33 [11,411ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-04-01 17:01:33 [11,414ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2024-04-01 17:01:33 [11,416ms] [Warning] [carb.audio.context] 1 contexts were leaked
2024-04-01 17:01:33 [11,455ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2024-04-01 17:01:33 [11,472ms] [Warning] [omni.core.ITypeFactory] Module /home/pcgadmin/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```

### System Info

Describe the characteristic of your environment:

- Commit: 3acff1bef9107ba6e82c9eec3cba424553ee1cfa
- Isaac Sim Version: 2023.1.1
- OS: Ubuntu 20.04
- GPU: RTX A6000
- CUDA: 12.0
- GPU Driver:525.147.05
### Additional context

This problems seems to be stemming from the recent update the omni.isaac.orbit.app Applauncher implementation 

### Checklist

- [ x] I have checked that there is no similar issue in the repo (**required**)
- [ x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ]  carb.settings.get_settings().get_as_string("/app/folder")  in "omni.isaac.version" version.py needs to return `/home/pcgadmin/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/` and not `/home/pcgadmin/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/../`

jsmith-bdai, 2024-04-01:
Thank you for posting! We are working on fixing this as it was introduced a recent commit integrating experience files. 

To get over this you can revert: https://github.com/NVIDIA-Omniverse/orbit/commit/fcc216a5399661b6f3125e1e0a99235f073a3124

Will post back when we've fixed the underlying issue - sorry for any inconvenience!

Ltesfaye, 2024-04-01:
Thank you for letting me know! I believe I've caught and fixed this bug with the pull request I just submitted. I didn't really have hte time to run though all the test cases but I believe it should be working. Hope that helps! 

Side note: 
This bug seems to be coming from the .kit files  in source/app pointing to a directory higher than they should be. It currently seems to be working with that change added.

hhansen-bdai, 2024-04-02:
@Ltesfaye, [dfa7487](https://github.com/NVIDIA-Omniverse/orbit/commit/dfa74872dd481167ea4740a4027a35985a3aea01) should solve the first problem and I think the second as well (I was able to instantiate a `LidarRtx` and its data path was under /isaac-sim/.) 

If you find that the problem persists, please post a script to reproduce and the Isaac Sim log in a new issue and I will be happy to make a fix. I'll close this issue. Thanks for your help! :smile:

Mayankm96, 2024-04-03:
@hhansen-bdai I looked at the commit you pointed and the PR marked here #340. I think the MR here seems like a more correct fix. 

The VERSION file Isaac Sim is loading is the one in Orbit right now and not the one in `_isaac_sim` directory.

commit 9eec9a27 (2024-04-04): Fixes`VERSION` file not found when launching Isaac Sim (#340)

## Description

The AppLauncher has a bug where it fails to fact the `app.folder` is
pointing to a directory higher than it should be. This leads to a file
not being found in the `_get_version` call from `omni.isaac.version`
module.

Updated app.folder to point to `"${exe-path}/"` instead of
`"${exe-path}/../"` in the following files:

- orbit/source/apps/orbit.python.headless.kit
- orbit/source/apps/orbit.python.headless.multicam.kit
- orbit/source/apps/orbit.python.kit

Fixes #339

## Type of change

- Bug fix (non-breaking chang [... cut]
commit ae222ba1 (2024-04-04): Fixes `VERSION` file not found when launching Isaac Sim (#340)

The AppLauncher has a bug where it fails to fact the `app.folder` is
pointing to a directory higher than it should be. This leads to a file
not being found in the `_get_version` call from `omni.isaac.version`
module.

Updated app.folder to point to `"${exe-path}/"` instead of
`"${exe-path}/../"` in the following files:

- orbit/source/apps/orbit.python.headless.kit
- orbit/source/apps/orbit.python.headless.multicam.kit
- orbit/source/apps/orbit.python.kit

Fixes #339

- Bug fix (non-breaking change which fixes an issue)

- [x] I h [... cut]
closed 2024-04-04 by Mayankm96