isaac-sim/IsaacLab #342: [Bug Report] Not Able to Launch Isaac Sim through Orbit

zoctipus, 2024-04-02:
If you are submitting a bug report, please fill in the following details and use the tag [bug].
after install following the documentation:
```python source/standalone/tutorials/00_sim/create_empty.py```
```./orbit.sh -p source/standalone/tutorials/00_sim/create_empty.py```
below bug is encountered, and isaac sim refuse to launch.
```
(tyorbit) octipus@octipus:~/Projects/tyorbit$ python source/standalone/tutorials/00_sim/create_empty.py
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /home/octipus/Projects/tyorbit/source/apps/orbit.python.kit
Starting kit application with the following args:  ['/home/octipus/Projects/tyorbit/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/octipus/Projects/tyorbit/source/apps/orbit.python.kit', '--/app/tokens/exe-path=/home/octipus/Projects/tyorbit/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/octipus/Projects/tyorbit/_isaac_sim/exts', '--ext-folder', '/home/octipus/Projects/tyorbit/_isaac_sim/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/octipus/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/logs/Kit/Isaac-Sim/2023.1/kit_20240402_112211.log
2024-04-02 18:22:11 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.018s] Failed to solve some dependencies locally, syncing with extension registry...
[0.020s] [ext: omni.kit.async_engine-0.0.0] startup
[0.021s] [ext: omni.assets.plugins-0.0.0] startup
[0.023s] [ext: omni.client-1.0.1] startup
[0.027s] [ext: omni.kit.registry.nucleus-0.0.0] startup
2024-04-02 18:22:39 [27,176ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[orbit.python-2023.1.1] dependency: (name: 'omni.kit.streamsdk.plugins', tag: '', version: '2.5.2', exact: 1, optional: 0) can't be satisfied. Available versions:
	 - [omni.kit.streamsdk.plugins-2.6.2+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.1+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.0+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.3.7+105.1.lx64.r] (/home/octipus/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.kit.streamsdk.plugins-2.3.7+105.1.lx64.r)
	 - [omni.kit.streamsdk.plugins-2.1.3+105.1.lx64.r] (/home/octipus/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.kit.streamsdk.plugins-2.1.3+105.1.lx64.r)
	 - [omni.kit.streamsdk.plugins-2.0.3+105.0.lx64.r] (registry)

2024-04-02 18:22:39 [27,176ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
```
I made sure the neuclus is installed, launched, and running,

I uninstalled and installed isaac sim 23.1.0 fresh, and downloaded orbit fresh, it seems non resolves this issue

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

- Isaac Sim Version: 2023.1.0
- OS: Ubuntu 22.04
- GPU: 4090
- CUDA: 12.2
- GPU Driver: 535.161.07

### Additional context

this issue first happened when I git pulled the newest repo

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

zoctipus, 2024-04-02:
I uninstalled the current system and reinstalled a Ubuntu 20.04 System, started everything fresh and the issue persists.....

zoctipus, 2024-04-02:
This issue is gone by switch to isaac sim 23.1.1

Mayankm96, 2024-04-03:
I have been seeing this also on some a different workstation. Have informed the team to look into it.

Which commit of Orbit messed things up for you? I made some changes in the app experience loading in fcc216a5399661b6f3125e1e0a99235f073a3124. I'm wondering if that caused the issues. It works fine on my PC though on another I see the issue.

zoctipus, 2024-04-03:
[fcc216a](https://github.com/NVIDIA-Omniverse/orbit/commit/fcc216a5399661b6f3125e1e0a99235f073a3124)
this commit I updated to caused the issue, specifically, if I look into `source/app/orbit.python.kit` I see 
line 21: `app.version = "2023.1.1"`
line 37: `"omni.kit.streamsdk.plugins" = {version = "2.5.2", exact = true}` 

These two line made me wanted to give a try installing isaac 23.1.1, So looking at the error message again when I had Isaac 1.0 :
```
2024-04-02 18:22:39 [27,176ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[orbit.python-2023.1.1] dependency: (name: 'omni.kit.streamsdk.plugins', tag: '', version: '2.5.2', exact: 1, optional: 0) can't be satisfied. Available versions:
	 - [omni.kit.streamsdk.plugins-2.6.2+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.1+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.0+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.3.7+105.1.lx64.r] (/home/octipus/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.kit.streamsdk.plugins-2.3.7+105.1.lx64.r)
	 - [omni.kit.streamsdk.plugins-2.1.3+105.1.lx64.r] (/home/octipus/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.kit.streamsdk.plugins-2.1.3+105.1.lx64.r)
	 - [omni.kit.streamsdk.plugins-2.0.3+105.0.lx64.r] (registry)
```
my guess it that when I installed isaac 23.1.0, I had steamsdk of all these version but not 2.5.2, maybe it only comes with isaac 23.1.

Mayankm96, 2024-10-04:
We fixed the streaming client versions in: https://github.com/isaac-sim/IsaacLab/pull/954

Please reopen this issue if the problem still persists. Thanks!

commit 477cd6b3 (2024-02-08): Adds entity clearing in scene for proper environment closing (#380)

# Description

The articulations, objects, and sensors were not cleared properly when
closing the environment. As a result, the invalidation callbacks were
not called, leading to problems if you want to open another environment
right after closing one.

Fixes #342

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./orbit.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes gener [... cut]
closed 2024-10-04 by Mayankm96