isaac-sim/IsaacLab #344: [Question] Control joint velocity of Franka Panda arm with robot.set_joint_velocity_target()

YZhang001, 2024-04-04:
### Question

Hello, I was following the tutorial on Orbit about the differential IK controller ([Link](https://isaac-orbit.github.io/orbit/source/tutorials/05_controllers/run_diff_ik.html)). It uses the joint position control command `robot.set_joint_position_target(joint_pos_des, joint_ids=robot_entity_cfg.joint_ids)` to set joint position target of a Franka Panda arm for each control loop (line 176). 

For my application I will need a velocity controller. I found `set_joint_velocity_target` in `source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py`. 
So I changed the joint position control command in `run_diff_ik.py` to the joint velocity control command `robot.set_joint_velocity_target(torch.zeros([1, 7], device=sim.device), joint_ids=robot_entity_cfg.joint_ids)` and `robot.set_joint_velocity_target(torch.ones([1, 7], device=sim.device), joint_ids=robot_entity_cfg.joint_ids)` to test whether the joints stay still at the default configuration (`torch.zeros()`) and whether all the joints rotate with a certain speed (`torch.ones()`). 

However, with the all-zero joint velocity input, the robotic arm does not stay still at the default configuration, but moves toward a configuration where it becomes to "stand upright", as shown in the following screenshot: 
[image]
Similarly, with the all-one joint velocity input, the robotic arm doesn't act as expected (all joints rotate at a certain speed), but stops at some state near the zero-configuration: 
[image]

I'm confused about this velocity command, because it does not seem to drive the joint towards a certain joint velocity, but to a certain configuration (joint position). I'm writing to ask if this is a common issue that still needs to be resolved or if I wrongly understand this function. Thank you!

pascal-roth, 2024-04-08:
@Mayankm96 any idea?

Mayankm96, 2024-04-12:
Are the PD gains are turned accordingly to do velocity control? I think they have a non-zero proportional component which makes it go to 0 (i.e. zero joint position).

YZhang001, 2024-04-15:
> Are the PD gains are turned accordingly to do velocity control? I think they have a non-zero proportional component which makes it go to 0 (i.e. zero joint position).

Hi @Mayankm96 thank you for your reply! Do you mean the stiffness and damping of franka configuration `FRANKA_PANDA_HIGH_PD_CFG` in `/source/extensions/omni.isaac.orbit_assets/omni/isaac/orbit_assets/franka.py`? I do see these PD gains in the configuration: 
`FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_shoulder"].stiffness = 400.0
FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_shoulder"].damping = 80.0
FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_forearm"].stiffness = 400.0
FRANKA_PANDA_HIGH_PD_CFG.actuators["panda_forearm"].damping = 80.0`
but the parameters themselves don't seem to distinguish between position control `set_joint_position_target` and velocity control `set_joint_velocity_target`. 
I tried to set both stiffness parameters to .0 and the issue mentioned above didn't happen again (the robot no more tried to get to the zero-joint position). But then it appears to me that the stiffnesses are defined for joint position error, dampings are defined for joint velocity error, which means there is only a damping parameter `FRANKA_PANDA_HIGH_PD_CFG.actuators[].damping` working for the `set_joint_velocity_target` command: this is equivalent to a P-controller for joint velocity (without derivative term for joint velocity) and the performance doesn't seem that reliable as a, for example, PD controller (k_p*velocity+k_d*acceleration). 

I would like to know whether there is a configuration from which I could get a PD controller for joint velocity control?