isaac-sim/IsaacLab #346: [Question] Cuda Error running self.robot.root_physx_view.get_jacobians() together with instance segmentation of camera

YitianShi, 2024-04-05:
### Question

Hi, I'm setting up a robot bin picking simulation with top-down camera that captures semantic or instance segmentation after each grasp attempt. I'm using my My simulation always get `CUDA error: an illegal memory access was encountered` when the jacobian of my robot is got from simulation:

```
self.robot.root_physx_view.get_jacobians()
```
Other modalities of camera such as rgb, normals and depths are working fine, while only the semantic or instance segmentation will cause such crash.

The similar issue that I found is:
https://forums.developer.nvidia.com/t/multiple-isaac-sim-containers-on-one-gpu-fails-with-cuda-illegal-memory-access-in-omni-physx-tensors-plugin/268134

Where I'm also sure that my GPU memory is far enough than exhaustion since I'm using 4 RTX4090GPUs to run only 3 bin-picking environments with only 2 objects in the bin.

pascal-roth, 2024-04-08:
As I understood, your environments do not require the multi-GPU setup. Could you try running the code on a single GPU to see if the same error still occurs? 
In any case, this error seems to be related to Isaac-Sim and TensorAPI instead of Orbit itself. To get better feedback, I would recommend to pose your questions in their channel.

Dhoeller19, 2024-10-05:
Hi @YitianShi,
Could you try this again with the latest Isaac Lab and Sim?

Holiclife-KTH, 2024-11-20:
[image]
@pascal-roth , @Dhoeller19 
I have similar problem with the latest Isaac Lab and Sim.
How can I fix it?
Thank you

RandomOakForest, 2025-01-04:
Thank you for posting this. @Holiclife-KTH, could you get a full stack trace and post this in a new issue if you still have this problem? This original question is too old and we may not be able to repro in the latest release. Please be sure to use Isaac Sim 4.2 and Isaac Lab 1.4.0. Thank you.

closed 2025-01-04 by RandomOakForest