isaac-sim/IsaacLab #347: [Bug] Gymnasium VideoRecorder class close the RlTaskEnv when closing one episode of video rendering

zoctipus, 2024-04-05:
Gymnasium VideoRecorder class close the RlTaskEnv when closing one episode of video rendering

Below code from
`gymnasium/wrappers/monitoring/video_recorder.py` 
```
def close(self):
        """Flush all data to disk and close any open frame encoders."""
        if not self.enabled or self._closed:
            return

        # First close the environment
        self.env.close()

        # Close the encoder
        if len(self.recorded_frames) > 0:
            try:
                from moviepy.video.io.ImageSequenceClip import ImageSequenceClip
            except ImportError as e:
                raise error.DependencyNotInstalled(
                    "moviepy is not installed, run `pip install moviepy`"
                ) from e

            clip = ImageSequenceClip(self.recorded_frames, fps=self.frames_per_sec)
            moviepy_logger = None if self.disable_logger else "bar"
            clip.write_videofile(self.path, logger=moviepy_logger)
        else:
            # No frames captured. Set metadata.
            if self.metadata is None:
                self.metadata = {}
            self.metadata["empty"] = True

        self.write_metadata()

        # Stop tracking this for autoclose
        self._closed = True
```
`self.env.close()` calles RlTaskEnv's close function and delete all the managers of the RlTaskEnv. (e.g. self.action_managers will become non-existent)

It seems simply comment out the ` self.env.close()` in above code resolves issue, but it is not recommended to change the source code, so I'm reporting here maybe seek for a better solution.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2024-04-05:
Can you check which Gymnasium version you have? You need to use 0.29.0.

zoctipus, 2024-04-06:
Yes thank you for reminding me this, I updated from 0.28.1 to 0.29.0 and the issue is gone

closed 2024-04-06 by zoctipus