isaac-sim/IsaacLab #352: [Question] Use ros2 in container

ruoshiwen, 2024-04-09:
### Ros2 support

Hi Orbit team,
I took a look in the docker-composer.yaml and there's a orbit-ros2 container defined. I run `./container.sh start`, but I could only see exisiting images: isaac-sim, orbit-base and ubuntu. My guess is that the image orbit-ros2 is not included in the container.sh file so not built.

I am very keen on using ros2 with orbit. It will be great if you could provide some instructions on how to enable this in the container.

Many thanks!

jsmith-bdai, 2024-04-09:
Hello! To start the ROS2 container, run `./container.sh start ros2`. For more info see this section: https://isaac-orbit.github.io/orbit/source/deployment/docker.html#orbit-image-extensions

ruoshiwen, 2024-04-09:
Yes, yes. I found that documentation right after I submitted this issue. Thank you very much, I will close it.

closed 2024-04-09 by ruoshiwen