isaac-sim/IsaacLab #358: [Question] Joint names and Body names ordering

hojae-io, 2024-04-12:
Hello everyone.

I'm looking into the ordering of the `joint_names` and `body_names` in `articulation.py`

It seems the name ordering is based on the hierarchical ordering of the URDF rigid body linking. 

For instance, when I call `asset.data.joint_names`, 
I get 
`['a01_right_hip_yaw', 'a06_left_hip_yaw', 'a02_right_hip_abad', 'a07_left_hip_abad', 'a03_right_hip_pitch', 'a08_left_hip_pitch', 'a04_right_knee', 'a09_left_knee', 'a05_right_ankle', 'a10_left_ankle']`

However, the ordering I want is alphabetical ordering such as
`['a01_right_hip_yaw', 'a02_right_hip_abad','a03_right_hip_pitch', 'a04_right_knee', 'a05_right_ankle', 'a06_left_hip_yaw', 'a07_left_hip_abad', 'a08_left_hip_pitch', 'a09_left_knee',  'a10_left_ankle']`

So I looked to the `articulation.py` where I found these lines of codes where I cannot dig into anymore.
[image]

My questions are 
1. What is the convention of orbit to determine the ordering of the joints and bodies?
2. Is there anyway that I can modify the code so that articulation interpret joint_names and body_names in alphabetical order?

I do not want to explicitly convert the ordering using another function, since I want all the related properties (stiffness, damping, inertia, friction etc..) to be ordered accordingly.

Thank you so much!

Mayankm96, 2024-04-18:
Hi @hojae-io ,

1. We use the convention followed by the PhysX stage parser for ordering bodies and joints. This convention is based on the articulation tree's breadth-first parsing.
2. No, currently, this is not possible directly with the simulator. We can relay this information to the PhysX team as a feature request. However, it won't be something available till later in the year.

hojae-io, 2024-04-18:
Yeah that would be appreciated. Thanks!