isaac-sim/IsaacLab #364: [Question] Filtering contacts between bodies

shafeef901, 2024-04-18:
### Question

Currently, the contact sensor works well for checking the contact forces on various links using a prim path expression in the form of regex. For example, 

`ContactSensorCfg(prim_path="{ENV_REGEX_NS}/Robot/.*", history_length=3, track_air_time=True)
`

Is it possible to filter out the collision/contacts of the robot with only certain objects in the environment? Currently, there exists an argument for the `ContactSensorCfg` called `filter_prim_paths_expr` which is supposed to do this, but when I pass expressions such as `/World/envs/env_.*/Object/link_.*`, it throws out the error:

`[Error] [omni.physx.tensors.plugin] Filter pattern '/World/envs/env_.*/Object/link_.*' did not match the correct number of entries`.

The Contact Sensor uses the [RigidContactView](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.prims.RigidContactView) class from `omni.isaac.core` but the example in the documentation only shows contact filtering with one environment. Using a regex expression instead in the same example doesn't seem to work and returns a similar error as above.

Since the documentation is lacking for this use case, can the authors provide any guide to perform filtering contacts using regex expressions?
 
Thanks in advance :)

Mayankm96, 2024-04-18:
Unfortunately, the filtering reports work properly between one rigid body prim to many filter prim paths.

So instead of providing: `"{ENV_REGEX_NS}/Robot/.*"`, you'll need to provide a specific body. For instance, `"{ENV_REGEX_NS}/Robot/RH_FOOT"`. I think the same holds for the filter prim paths (where you'd need to manually list them instead of having `link_.*`). But you'll have to check that up.

Let me know if this fixes the issue for you!

shafeef901, 2024-04-19:
Using "{ENV_REGEX_NS}/Robot/.*" for the prim path works, as explicated in the Orbit Quadruped examples where it is being used to detect foot and base contacts with the ground. But unfortunately, it is when using a similar expression in the filter prim path, the error pops up.

Mayankm96, 2024-04-19:
Yes, but can you make a separate contact sensor for each part of the robot and see if filtering works then?

```python
lh_foot_contact_forces = ContactSensorCfg(prim_path="{ENV_REGEX_NS}/Robot/LH_FOOT", history_length=3, track_air_time=True, filter_prim_paths=["/World/envs/env_.*/Object/link_.*])
base_contact_forces = ContactSensorCfg(prim_path="{ENV_REGEX_NS}/Robot/base", history_length=3, track_air_time=True, filter_prim_paths=["/World/envs/env_.*/Object/link_.*])
```

shafeef901, 2024-04-22:
Unfortunately not, it still throws up the error for num_envs = 2:
`[Error] [omni.physx.tensors.plugin] Filter pattern '/World/envs/env_.*/Object/link_.*' did not match the correct number of entries (expected 2, found 0)`

shafeef901, 2024-04-26:
The contact filtering works by explicitly adding every link of every environment as below. However, this is highly redundant and probably makes it slower.

```python
filter_prim_paths=[f"/World/envs/env_{i}/Object/link{j}" for i in range(num_envs) for j in range(num_links)]
```

Mayankm96, 2024-04-26:
> The contact filtering works by explicitly adding every link of every environment as below. However, this is highly redundant and probably makes it slower.

It shouldn't make it slower since regex expressions are resolved internally by PhysX in the same way. If you only care about interaction with particular links of the object (i.e. you know all of them don't make sense), you should remove them.

Mayankm96, 2024-04-26:
Maybe we can fix the documentation on this to avoid confusion for future?

shafeef901, 2024-04-26:
It seems to me that filtering doesn't work with regex expressions. So `filter_prim_paths_expr` should probably be renamed to `filter_prim_paths`. Furthermore, since we define `contact_sensor` in the cfg file, it's not straightforward to infer `num_envs`.  In my case, I hardcoded it to test it. Is there a better way to do so?

Mayankm96, 2024-04-26:
Ah wait. It doesn't work for `/World/envs/env_*/Object/link0`?

Mayankm96, 2024-04-26:
@shafeef901 Can you check the MR: https://github.com/NVIDIA-Omniverse/orbit/pull/385?

closed 2024-04-27 by Mayankm96

AravindhPadmanabhan, 2024-05-13:
Hello @Mayankm96 ,

I think I found a mistake in the newly added contact force assertions used in the unit test for the contact sensor.

I tried running [test_contact_sensor.py](https://github.com/NVIDIA-Omniverse/orbit/pull/385/files#diff-86fd023c077adbc6d6d9752244fd2a51e8381e209b9aebebcb300cdde0715e3e) and printed out the tensors being used in the assertions in line 292 and 295 and they turn out to be all 0s. Looks like the assertion passes only because they are all zero tensors.

I came across this when I was trying to find the contact forces and contact points of a robot with the ground. I tried this out using the example in `source/standalone/tutorials/04_sensors/add_sensors_on_robot.py`. I was able to get the contact forces using `ContactSensor.data.net_forces_w`. But to get contact points, I specified filters in `ContactSensorCfg.filter_prim_paths_expr` and used `ContactSensor.contact_physx_view.get_contact_data()`. Hence, I modified the `ContactSensorCfg` to look like this:

```
contact_forces = ContactSensorCfg(
        prim_path="{ENV_REGEX_NS}/Robot/.*_FOOT", update_period=0.0, history_length=6, debug_vis=True,
        filter_prim_paths_expr = ["/World/defaultGroundPlane"]
    )
```

I printed out `Contact_sensor.data.net_forces_w`, `ContactSensor.data.force_matrix_w` and `ContactSensor.contact_physx_view.get_contact_data()`. While net_forces_w had non-zero values which seemed reasonably correct, `force_matrix_w` and `get_contact_data()` were filled with 0s, as shown in the image below. I also got a warning that said `GPU contact filter for collider '/World/defaultGroundPlane' is not supported`. Could you please look into this and let me know how I can get the contact points between two prims? Thanks in advance!

[image]

P.S.: Also had to make `max_contact_data_count` a non-zero value in the `create_rigid_contact_view()` function in line 258 of `source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/contact_sensor/contact_sensor.py` for `get_contact_data()` to work

xzhu0428, 2024-06-04:
@AravindhPadmanabhan Based on [this](https://github.com/isaac-sim/IsaacLab/blob/3ad18a8e1a5c166ad1a22f105d47a5c578de68d7/source/extensions/omni.isaac.lab/omni/isaac/lab/terrains/utils.py#L91) and [this](https://github.com/isaac-sim/IsaacLab/blob/3ad18a8e1a5c166ad1a22f105d47a5c578de68d7/source/extensions/omni.isaac.lab/omni/isaac/lab/terrains/terrain_importer.py#L89), using `filter_prim_paths_expr=["/World/defaultGroundPlane/terrain/mesh"]` should work.

AravindhPadmanabhan, 2024-06-19:
@xzhu0428 , thanks a lot, your suggestion worked and I am able to get contact points. 

But, I still have to specify max_contact_data_count (in line 260 of contact_sensor.py) to a non-zero value and if I set it to a value more than the number of bodies, I get more contact points and normals than the number of bodies in the contact sensor. And I get the following warning:
`Incomplete contact data is reported in GpuRigidContactView::getContactData because there are more contact data points than specified maxContactDataCount = 24.
`
Could you please enlighten me on what these contact points mean and guide me to some documentation on how to find out which row corresponds to which body?

PerFrivik, 2024-07-11:
Thank you @xzhu0428, using:

`contact_forces = ContactSensorCfg(prim_path="{ENV_REGEX_NS}/Robot/.*",filter_prim_paths_expr =["{ENV_REGEX_NS}/Robot/LF_FOOT", ... "**/World/ground/terrain/mesh**"], history_length=3, track_air_time=True) `

worked for me!

PerFrivik, 2024-07-11:
> @xzhu0428 , thanks a lot, your suggestion worked and I am able to get contact points.
>
> But, I still have to specify max_contact_data_count (in line 260 of contact_sensor.py) to a non-zero value and [...]

I had the same issue when I was working with the rough terrain ground mesh, but when I tried a flat environment using the ground plane: 

`/World/ground/GroundPlane/CollisionPlane`

I did not get this issue, this collision plane didnt show up as a mesh, which leads me to believe that the mesh might have more collisions? I'm really not sure thought, I'll let you know once I find out more!

xliu0105, 2024-07-23:
> @AravindhPadmanabhan Based on [this](https://github.com/isaac-sim/IsaacLab/blob/3ad18a8e1a5c166ad1a22f105d47a5c578de68d7/source/extensions/omni.isaac.lab/omni/isaac/lab/terrains/utils.py#L91) and [...]

After I set it up like this, I encountered this error: 2024-07-23 10:58:51 [9,486ms] [Error] [omni.physx.tensors.plugin] Filter pattern '/World/GroundPlane/terrain/mesh' did not match the correct number of entries (expected 10, found 0)

zxhuang97, 2024-10-01:
> After I set it up like this, I encountered this error: 2024-07-23 10:58:51 [9,486ms] [Error] [omni.physx.tensors.plugin] Filter pattern '/World/GroundPlane/terrain/mesh' did not match the correct [...]

@xliu0105 For the ground plane, you may configure the ContactSensor as:
```    
contact_forces = ContactSensorCfg(
        prim_path="{ENV_REGEX_NS}/Robot/LF_FOOT",
        filter_prim_paths_expr=["/World/defaultGroundPlane/GroundPlane/CollisionPlane"],
    )
```

However, I still have similar issues as @shafeef901 when I try to filter multiple prims.
```
prim_path="{ENV_REGEX_NS}/Object", 
#  doesn't work
filter_prim_paths_expr= ["{ENV_REGEX_NS}/Robot/.*finger"],
#  works
filter_prim_paths_expr= ["{ENV_REGEX_NS}/Robot/.*rightfinger", "{ENV_REGEX_NS}/Robot/.*leftfinger"], 
```
When using the the first expression, an error will pop out
```
[Error] [omni.physx.tensors.plugin] Filter pattern '/World/envs/env_*/Robot/*finger' did not match the correct number of entries (expected 1, found 2)
```
@Mayankm96 It seems to me that each expression can only correspond to one prim, is it correct?

Milad-Rakhsha-NV, 2024-12-11:
regardig `max_contact_data_count`: 
Specifying that is indeed required to find all contacts for all patches. However, just to clarify;  this is used to get individual contact forces between body pairs not only their aggregate, which is what is reported by the get_contact_force_matrix(). Two bodies in contact can report more than one contact point (e.g. imagine two boxes one sitting on top of the other). For this simple example, although you have 2 bodies, you are expected to specify a lower bound on the number of contacts being generated with max_contact_data_count (4-10 contact data point should be enough for this example). 
Therefore, max_contact_data_count should be large enough to allow storing all the contact points, and it is up to the user to decide how to process all the information. We cannot know in advance the type of contact interactions users will try so specifying this parameter is left to the users.
There are good examples on how to use the underlying physics_view in this extension `omni.physics.tensors.tests` under `python/scripts/tests.py`, e.g. `TestRigidContactMatrix` which I recommend.

boocoly, 2024-12-17:
> @xzhu0428 , thanks a lot, your suggestion worked and I am able to get contact points.
>
> But, I still have to specify max_contact_data_count (in line 260 of contact_sensor.py) to a non-zero value and [...]

@AravindhPadmanabhan. I think I have understood the parameters, including the return of `get_contact_data` and `max_contact_data_count`.

for `data = get_contact_data(dt=dt)`, I think the most important value is  `data[4]` and `data[5]`. If `num_envs =10`, `num_bodies=5`, and the filter_prim is just "/World/defaultGroundPlane/GroundPlane/CollisionPlane", then the data[4] and data[5] will always have shape `(50,1)`, because there are 50 (num_envs $\times$ num_bodies) contact pairs, and each contact pair can have multiple contact points. 

data[4] and data[5] - Pair Contact Counts and Start Indices
- data[4] shows the number of contact points of each pair. Maybe some pairs  have 4 contact points (maybe their contact is a surface contact), others have 0, etc.
- data[5] provides starting indices for these contacts in the contact force arrays.
With data[4] and data[5], we can get every force and the contact point that we want.

`max_contact_data_count` influences the `data[0],data[1] ... data[3]`. The shape of these 4 tensors is [max_contact_data_count,-1]. For example, if there are just two cubes that have face-to-face contact, and we set `max_contact_data_count=10000`, then, there will be 10000 points to describe these face-to-face contact. And data[4] will equals tensor([[10000]]), data[5] will equals tensor([0]).

But I still have unclarity. What's the difference between one point description and multi point description for face-to-face contact? Does the one point contact represent the total force?

Please verify me if there is any wrong understanding.