isaac-sim/IsaacLab #365: [Question] How to fix base link.

john2choi, 2024-04-19:
Hi, Thank you for the nice work,

Main question: How to fix the base link of robot arm on the ground? 

I am using Techman tm5-900 with Tesollo delto-gripper (DG3F) generated by URDF Importer2usd.
When I imported the robot's URDF, unchecked the "Fix Base Link" button as an import option because It couldn't generate multiple environments (can generate but all robots are fixed to the base link) in RLTaskEnv.

Could I fix the base link of each robot base link to the ground when I run the model?

Thank you already.

(You can check the situation.)
[image]

Mayankm96, 2024-04-19:
>  It couldn't generate multiple environments (can generate but all robots are fixed to the base link) in RLTaskEnv.

Why is this the case? Isn't it what you want?

john2choi, 2024-04-21:
Hi,

In simulation rendering, setting the robot as a fixed base link means that it becomes fixed to the base frame as you can see below. Therefore, the question was whether there is a way to fix the base link in each environment during the envs post-process.

Thank you for the answer.

[image]

Mayankm96, 2024-04-21:
It should be possible to move the robot's base to different origins: https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/standalone/tutorials/02_scene/create_scene.py

john2choi, 2024-04-30:
Solved with https://github.com/NVIDIA-Omniverse/orbit/pull/367

closed 2024-04-30 by john2choi