isaac-sim/IsaacLab #366: [Bug Report] DifferentialIKController not working correctly when applying rotation to Articulation

samueledelloca, 2024-04-19:
### Describe the bug

I'm developing a custom environment which makes (also) use of the `DifferentialIKController` for reaching a specific pose, defined through a command. Everything works fine until I apply a rotation into the Articulation's `init_state`. It took me a while to understand the cause, because the Articulation's behavior is pretty strange. See the attached example videos for both my custom environment and `Isaac-Lift-Cube-Franka-IK-Abs-v0` task from `orbit_tasks/manipulation/lift`. 
I tried with different articulations and configurations but the error persists, even when using only RL and `DifferentialInverseKinematicsAction` without state machine. Other actions are working fine when rotating the Articulation. So I suppose that the initial rotation is not taken into account when computing the change in joint positions through the IK controller, but I'm not sure how to fix it.

### Steps to reproduce

Simply add the following lines into `FRANKA_PANDA_HIGH_PD_CFG` and execute the `Isaac-Lift-Cube-Franka-IK-Abs-v0` task with `./orbit.sh -p source/standalone/environments/state_machine/lift_cube_sm.py --num_envs 8`.
Note that this task aims first at reaching the cube object and then lifting it to a specific pose. I'm interested only in the second part, so I didn't fixed the code for reaching the cube after applying changes to initial position and rotation.

### Additional context

You will find 2 attached videos: [`custom_environment`](https://github.com/NVIDIA-Omniverse/orbit/assets/34474980/3a4f5435-4501-4e81-af29-46f47066562e) (I apologize for the quality but it's the best I can upload) which shows the bug in my custom environment and [`franka_lift`](https://github.com/NVIDIA-Omniverse/orbit/assets/34474980/0a1581f3-3b13-4c91-a51e-a0e37fc0bab0) which is related to `Isaac-Lift-Cube-Franka-IK-Abs-v0` and can be reproduced following the instructions in the previous section. The latter video is divided into 3 parts: 
1- correct behavior without applying rotation 
2- wrong behavior by applying rotation and translation 
3- correct behavior by applying only translation 

While for the custom environment video you have:
1- correct behavior without applying rotation 
2- wrong behavior by applying rotation and translation 

### Checklist

- [X] I have checked that there is no similar issue in the repo
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Articulation behaving correctly (i.e. by reaching the defined pose) using `DifferentialIKController` when applying an initial rotation

Mayankm96, 2024-04-19:
Are the targets given in the same frame as in which the controller is working?

samueledelloca, 2024-04-20:
> Are the targets given in the same frame as in which the controller is working?

Yes, I'm pretty sure they are, given that the target frames are visualized correctly as you can see in the videos. They are even reached correctly if I do not apply any initial rotation to the Articulation.

AntoineRichard, 2026-08-27:
Thanks for the detailed write-up and video, @samueledelloca — rotating the base is a great stress test for the IK term.

The controller itself does handle a rotated root: `DifferentialInverseKinematicsAction._compute_frame_pose` applies `subtract_frame_transforms(root_pos_w, root_quat_w, ...)` and the Jacobian is rotated by `quat_inv(root_quat_w)` (`source/isaaclab/isaaclab/envs/mdp/actions/task_space_actions.py`). The demo is the mismatch: `scripts/environments/state_machine/lift_cube_sm.py` builds targets as `root_pos_w - scene.env_origins`, i.e. env-origin frame, which only equals the base frame for an unrotated robot at the origin.

Closing as a frame-convention issue. You also mentioned it reproducing in a custom RL env without the state machine — if that's still true on a current release, please reopen with that minimal repro and we'll take a look.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard