isaac-sim/IsaacLab #371: Why multiply by dt when calculating the reward?

swstbecrpgmail, 2024-04-21:
Why multiply by dt when calculating the reward, if I may ask politely?

[image]

Mayankm96, 2024-04-22:
From the documentation: https://isaac-orbit.github.io/orbit/source/api/orbit/omni.isaac.orbit.managers.html#reward-manager

> The reward manager multiplies the reward term’s weight with the time-step interval dt of the environment. This is done to ensure that the computed reward terms are balanced with respect to the chosen time-step interval in the environment.

swstbecrpgmail, 2024-04-22:
> From the documentation: https://isaac-orbit.github.io/orbit/source/api/orbit/omni.isaac.orbit.managers.html#reward-manager

Thank you very much! This clears up my confusion.

commit 8ff79600 (2024-02-22): Checks docker version before building singularity image  (#402)

# Description

For the cluster deployment, the latest docker version (25.xx) is not yet
compatible with the latest apptainer version. This PR adds a check
before building the singularity image and adds a warning to the docs.
Also the NOHTTPS warning has been resolved.

Fixes #371 

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- This change requires a documentation update

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./orbit.sh --format`
- [x] I have made corres [... cut]
closed 2024-04-22 by swstbecrpgmail