isaac-sim/IsaacLab #379: GUI rendering not in real-time

hojae-io, 2024-04-23:
Hi all,

I'm running the play script in my local computer (bare-metal) with cpu (Intel i-9 11th gen). 
I'm trying to run only 1 humanoid robot with `sim_dt=0.001` and `decimation=10`

```
[INFO]: Base environment:
        Environment device    : cpu
        Physics step-size     : 0.001
        Rendering step-size   : 0.001
        Environment step-size : 0.01
        Physics GPU pipeline  : False
        Physics GPU simulation: False
```

But I'm experiencing slow motion on my GUI rendering. 

From my past experience with IsaacGym, if I run the play script with `cpu` device, the GUI rendering was real-time speed. (using `cuda:0` is slower than using `cpu`)
However, now in Orbit, even with the `cpu` device, the rendering speed seems almost x 3 ~ x 5 times slower. 

Of course if I increase sim_dt and lower decimation (e.g. `sim_dt=0.005`, `decimation=4`), then the rendering gets faster. 

I want to keep my original params (`sim_dt=0.001` and `decimation=10`) while rendering in real-time speed.

Is there a way that I can get the real-time speed? Or is it just bottleneck by the speed of the physx and renderer?

Thank you!

mikelasa, 2024-05-16:
I'm interested on this too, I'm using the same params and experiencing a slow motion simulation

LyuJZ, 2024-06-11:
I used the latest IsaacLab, and found that the self.sim.render() takes too long (5 seconds). Any comments?

kellyguo11, 2024-09-27:
Isaac Lab rendering performance could also depend on the GPU hardware available on the machine. The renderer in Isaac Gym was quite simple and didn't have ray tracing capabilities, while the renderer in Isaac Lab requires a lot more compute. You could try adjusting some of the render setting parameters from the UI to lower the rendering quality a bit, which could help with performance. Some settings we use for rendering could be found [here](https://github.com/isaac-sim/IsaacLab/blob/main/source/apps/isaaclab.python.headless.rendering.kit#L41).

closed 2024-09-27 by kellyguo11