isaac-sim/IsaacLab #387: [Question] Cylindrical wheels in orbit vs. Isaac Sim

MaltheT, 2024-04-29:
### Cylindrical wheels in orbit vs. Isaac Sim

Hi,

I'm using the orbit to simulate a robot with cylindrical wheels. I've noticed that the wheels' colliders are rougher than expected, leading to bumpy driving behavior, which does not occur when driving the robot in Isaac Sim. 

I expected the wheels to handle minor surface variations smoothly, but instead, the driving is noticeably bumpy.

Is this a known issue, and could you suggest any adjustments to the collider settings or other workarounds?

Isaac Sim colliders:
[image]

Orbit colliders:
[image]

Thank you for your assistance!

Mayankm96, 2024-04-29:
Can you try setting these to True and see if this solves the issue?

> https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py#L133-L137

MaltheT, 2024-04-30:
Yes that fixed it, thank you!

commit 83d62e21 (2024-03-21): Fixes imitation learning workflow for lift environment (#451)

# Description

This PR fixes the imitation learning workflow in that
`collect_demonstrations`, `train` and `play` scripts all don't throw
exceptions. I haven't validated that the training actually generates a
successful policy, only that the loss decreased within the first few
iterations.

A follow up task might be to make sure that the chosen observation terms
can still result in a good policy, but that's outside of the scope of
updating this workflow to API changes in Orbit and Robomimic.

Fixes #387 

## Type of change

- Bug fi [... cut]
closed 2024-04-30 by MaltheT