isaac-sim/IsaacLab #390: [Bug Report] Container install failed

swstbecrpgmail, 2024-04-30:
When I run ./docker/container.sh on Ubuntu 20.04, it reports that:  
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
15.21 nvidia-srl-usd 0.13.0 requires tqdm<5.0.0,>=4.63.0, which is not installed.
15.21 nvidia-srl-usd 0.13.0 requires usd-core<24.00,>=21.11, which is not installed.

Sorry to bother you again, but I truly wonder why it occurs. I would greatly appreciate it if you could help me solve this problem or give some potential solutions.

[image]
[image]
[image]
[image]

swstbecrpgmail, 2024-05-04:
Additionally, the docker version is 26.1.1, and the docker-compose version is 1.25.0.

MuhongGuo, 2024-05-06:
The issue can be workaround if I add a step in `Dockerfile.base` to install usd-core. See:

```
# Install usd-core 
RUN ${ORBIT_PATH}/orbit.sh -p -m pip install usd-core==23.11

# installing Orbit dependencies
# use pip caching to avoid reinstalling large packages
RUN --mount=type=cache,target=${DOCKER_USER_HOME}/.cache/pip \
    ${ORBIT_PATH}/orbit.sh --install --extra
```

swstbecrpgmail, 2024-05-06:
> The issue can be workaround if I add a step in `Dockerfile.base` to install usd-core. See:
> 
> ```
> # Install usd-core 
> RUN ${ORBIT_PATH}/orbit.sh -p -m pip install usd-core==23.11
> 
> # installing Orbit dependencies
> # use pip caching to avoid reinstalling large packages
> RUN --mount=type=cache,target=${DOCKER_USER_HOME}/.cache/pip \
>     ${ORBIT_PATH}/orbit.sh --install --extra
> ```
> 
> cc @Mayankm96

Thank you very much! This really solves some errors for me! But one error relative to robomimic still remains: 
[image]
I will true appreciate it if anyone has potential solutions or tell me the possible reasons for this error.

MuhongGuo, 2024-05-07:
@swstbecrpgmail I think it's a network issue when cloning robomimic. I also had this issue but a retry can resolve it. 

In case you don't need `robomimic` in your project, you can modify `Dockerfile.base` to only install other dependencies: `${ORBIT_PATH}/orbit.sh --install --extra rl_games,rsl_rl,sb3`.

If you do need `robomimic`, you may refer to this [solution](https://stackoverflow.com/a/6849424) and test it with this command in a new terminal: `git clone https://github.com/ARISE-Initiative/robomimic.git`.

MuhongGuo, 2024-05-08:
@swstbecrpgmail Well it seems like the "usd-core missing" issue doesn't cause any problems for the usages. You may just follow the above suggestions to resolve you docker build problem and this github issue can be kindly closed I think.

swstbecrpgmail, 2024-05-25:
> @swstbecrpgmail I think it's a network issue when cloning robomimic. I also had this issue but a retry can resolve it.
>
> In case you don't need `robomimic` in your project, you can modify [...]

I apologize for not being able to respond to you in a timely manner due to being quite busy recently. Based on your reply, I just tested your solution. While I failed to install robomimic, I successfully installed orbit. Thank you again for your response!

Additionally, I couldn't run the demo script correctly in the lastest-version orbit, so I still choose to work on the old version.

Again, Thank you very much!

swstbecrpgmail, 2024-05-26:
Update! The reason for my installation failure is most likely due to the unstable network (I am a developer from China). If you encounter installation failures similar to the failures mentioned above, just try a few more times.

closed 2024-05-26 by swstbecrpgmail