isaac-sim/IsaacLab #391: [Question] How to control Articulation actuators independently.

SantiDiazC, 2024-05-01:
### Question

Hi, I am creating my scene on orbit and I define a robot using the ArticulationCfg class, such the joints of the robot are defined like this:

``` python
init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.0), 
        joint_pos={"l_ax0_joint_0": 0.0, 
                   "l_ax0_joint_1": 0.0,
                   "l_ax0_joint_2": 0.0,
                   "l_ax0_joint_3": 0.0,
                   "l_ax0_joint_4": 0.0,
                   "l_ax1_joint_0": 0.0,
                   "l_ax1_joint_1": 0.0,
                   "l_ax1_joint_2": 0.0,
                   "l_ax1_joint_3": 0.0,
                   "l_ax1_joint_4": 0.0,
                   },
    ),
    actuators={
        "l_ax0": ImplicitActuatorCfg(
            joint_names_expr=["l_ax0_joint_.*"],
            effort_limit=100.0,
            velocity_limit=100.0,
            stiffness=100.0,
            damping=10.0,
        ),
        "l_ax1": ImplicitActuatorCfg(
            joint_names_expr=["l_ax1_joint_.*"],
            effort_limit=100.0,
            velocity_limit=100.0,
            stiffness=100.0,
            damping=10.0,
        ),
        
    },

```
I want to control the two set of joints (actuators) independently  and I checked the documentation [Interacting With An Articulation](https://isaac-orbit.github.io/orbit/source/tutorials/01_assets/run_articulation.html),
but in that example all joints commands are applied at the same time without specifying any actuator, so I am curious if there is a way to do that.

Thank you in advance!

Mayankm96, 2024-05-01:
Hi @SantiDiazC ,

This is currently not possible through the tensor API themselves. Even if you set the command for a sub-set of the joints, the entire tensor (at the joint-indexing level) is set to the simulation. This means the targets would be zero for those you don't set into the buffers directly.

If you want to actuate only a sub-set of joints, there are two options:

1. Make the PD gains of the implicit actuators zero for the other joints -- This means PhysX will not compute any torques using the PD control law, and the "resultant" applied effort will be zero.
2. Set the command of the other joints equal to their current or fixed joint positions -- This means they will be commanded to stay where they are.

SantiDiazC, 2024-05-02:
Thank you for your reply!
Ok I understand, It would be great to have that kind of feature in the future so we can model more complex behaviors.

for the first case it means to set the implicit actuator config like this?:

``` python
"l_ax0": ImplicitActuatorCfg(
            joint_names_expr=["l_ax0_joint_.*"],
            effort_limit=100.0,
            velocity_limit=100.0,
            stiffness=0.0,
            damping=0.0,
```
Setting the stiffness (P gain) and damping (D gain) to 0.0? ([ImplicitActuatorCfg](https://isaac-orbit.github.io/orbit/source/api/orbit/omni.isaac.orbit.actuators.html#omni.isaac.orbit.actuators.ImplicitActuatorCfg)). Can it be modified while the simulation is running? or only at the initialization stage?