isaac-sim/IsaacLab #392: [Question] Some demo examples cannot be executed in Issac sim2023.1.1

17863958533, 2024-05-01:
Hello, while using the latest version of Orbeit program for testing at Issac sim2023.1.1, I found that some demo examples cannot be executed.
For example, when I execute a script according to a command:
./orbit.sh - p source/standalone/demos/quadrupeds. py
When running the above example, although the Isaac sim interface was successfully launched, no further actions could be taken. My error message is as follows:
-------------------------------------------------------------------------------------------------------------------------------
......
Active user not found. Using default user [kiosk]Streaming server started.
[2.933s] [ext: omni.kit.window.stats-0.1.4] startup
[2.934s] [ext: orbit.python-2023.1.1] startup
[2.935s] Simulation App Starting
2024-05-01 04:59:47 [2,942ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2024-05-01 04:59:47 [2,942ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2024-05-01 04:59:47 [2,942ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2024-05-01 04:59:47 [2,943ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2024-05-01 04:59:47 [2,943ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available
2024-05-01 04:59:47 [2,943ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2024-05-01 04:59:48 [3,772ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-05-01 04:59:48 [3,772ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[4.555s] app ready
2024-05-01 04:59:49 [4,800ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-05-01 04:59:49 [4,800ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[4.962s] RTX ready
[5.078s] Simulation App Startup Complete
[5.123s] [ext: omni.kit.stagerecorder.core-105.0.4] startup
[5.998s] [ext: omni.physx.fabric-105.1.12-5.1] startup
2024-05-01 04:59:54 [9,290ms] [Warning] [carb] Client omni.stageupdate.plugin has acquired [omni::hydra::IOmniHydra v2.0] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
2024-05-01 05:00:15 [30,817ms] [Warning] [omni.client.python] Detected a blocking function. This will cause hitches or hangs in the UI. Please switch to the async version:
  File "/home/pc/Orbit_lyy/orbit/source/standalone/demos/quadrupeds.py", line 183, in <module>
  File "/home/pc/Orbit_lyy/orbit/source/standalone/demos/quadrupeds.py", line 174, in main
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py", line 380, in reset
  File "/home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 624, in reset
  File "/home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 592, in initialize_physics
  File "/home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 889, in play
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py", line 451, in render
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py", line 92, in <lambda>
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py", line 255, in _initialize_callback
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 606, in _initialize_impl
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 703, in _process_actuators_cfg
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/actuators/actuator_net.py", line 50, in __init__
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/assets.py", line 146, in read_file
  File "/home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/extscore/omni.client/omni/client/__init__.py", line 491, in read_file

2024-05-01 05:07:12 [447,608ms] [Error] [carb.events.python] RuntimeError: istream reader failed: getting the current position.

At:
  /home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/jit/_serialization.py(164): load
  /home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/actuators/actuator_net.py(51): __init__
  /home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py(703): _process_actuators_cfg
  /home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py(606): _initialize_impl
  /home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py(255): _initialize_callback
  /home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py(92): <lambda>
  /home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py(451): render
  /home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(889): play
  /home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(592): initialize_physics
  /home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(624): reset
  /home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py(380): reset
  /home/pc/Orbit_lyy/orbit/source/standalone/demos/quadrupeds.py(174): main
  /home/pc/Orbit_lyy/orbit/source/standalone/demos/quadrupeds.py(183): <module>

2024-05-01 05:07:32 [467,622ms] [Warning] [omni.client.python] Detected a blocking function. This will cause hitches or hangs in the UI. Please switch to the async version:
  File "/home/pc/Orbit_lyy/orbit/source/standalone/demos/quadrupeds.py", line 183, in <module>
  File "/home/pc/Orbit_lyy/orbit/source/standalone/demos/quadrupeds.py", line 174, in main
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py", line 380, in reset
  File "/home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 624, in reset
  File "/home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 592, in initialize_physics
  File "/home/pc/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 889, in play
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py", line 451, in render
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py", line 92, in <lambda>
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py", line 255, in _initialize_callback
  File "/home/pc/Orbit_lyy/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 606, in _initialize_impl
[... 184 lines cut ...]
+-------------------------------+----------------------+----------------------+
[... cut ...]
Built on Thu_Nov_18_09:45:30_PST_2021
Cuda compilation tools, release 11.5, V11.5.119
Build cuda_11.5.r11.5/compiler.30672275_0

May I ask how my situation was caused?
I reinstalled the software and found that it still couldn't solve the problem. At the same time, I tested it in the Conda environment and the problem remained the same. I think this may be a bug, and I would greatly appreciate it if you could reply. My email is: 2310267@tongji.edu.cn It is also possible to reply below this chat, thank you very much.

Mayankm96, 2024-05-01:
Are you connected to Isaac Sim nucleus?

> https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_faq.html#assets-and-nucleus

17863958533, 2024-05-02:
Yes, although I did not download the three packages for Isaac sim offline, I believe that I have configured Nucleus and it works fine when I check the connection status in issac sim. I can also load the relevant USD model files on the interface normally by dragging them.
[image]

17863958533, 2024-05-03:
Okay, I have solved this problem and the methods I have taken are as follows:
(1) Firstly, search for "ISAAC SIM ASSETS PACK" in the Omnivers Launcher interface and download these three packages locally.
(2) Merge the downloaded three material packages (Material Package 1, 2, and 3) into Material Package 1. You can refer to the following command.

# Copy package 3 to package 1
cp -v -r ~/Downloads/isaac_sim-assets-3-2023.1.0/Assets/Isaac/2023.1.0/NVIDIA/* ~/Downloads/isaac_sim-assets-1-2023.1.0/Assets/Isaac/2023.1.0/NVIDIA/
# Copy package 2 to package 1
cp -v -r ~/Downloads/isaac_sim-assets-2-2023.1.0/Assets/Isaac/2023.1.0/NVIDIA/* ~/Downloads/isaac_sim-assets-1-2023.1.0/Assets/Isaac/2023.1.0/NVIDIA/
# Go to assets parent directory
cd ~/Downloads/isaac_sim-assets-1-2023.1.0/Assets/Isaac/2023.1.0/NVIDIA

(3) Confirm the specific path of the downloaded offline package in NUCLEUS and use the following command when starting Isaac sim from the terminal initially.
./isaac sim sh --/persistent/isaac/assetroot/default=“ omniverse://localhost/Library/NVIDIA/Assets/Isaac/2023.1.0 “
The above“ omniverse://localhost/Library/NVIDIA/Assets/Isaac/2023.1.0 I need to change to your local path, for example, my own local path is“ file:///media/pc/C8BE4D94BE4D7BC6/IsaacSim_Pack/isaac_sim-assets-1-2023.1.1/Assets/Isaac/2023.1.1 ”.
(4) Modify the program under the path "/home/pc/. local/share/ov/pkg/isaac_sim-2023.1.1", with a focus on modifying the script under the path "/home/pc/Orbit_lyy/orbit/_isaac_sim/exts/omni. isaac. core/omni/isaac/core/utils/nucleus. py" and locate line 563. The modified changes are as follows:
----------------------------------
......
# get timeout
timeout = carb.settings.get_settings().get("/persistent/isaac/asset_root/timeout")
if not isinstance(timeout, (int, float)):
timeout = 10.0
﻿
# 0 - Check local path first
carb.log_info("Checking local assets path")
local_path = "file:///media/pc/C8BE4D94BE4D7BC6/IsaacSim_Pack/isaac_sim-assets-1-2023.1.1/Assets/Isaac/2023.1.1" #This is your own path.
if os.path.exists(local_path.replace('file://', '')):  # Remove 'file://' for os.path.exists
carb.log_info("Assets root found locally at {}".format(local_path))
return local_path
﻿
# 1 - Check /persistent/isaac/asset_root/default setting
......
-----------------------------------

(5) Check that there are no issues with the Orbit environment under Conda, and run it again to successfully run the demo!

Wtrwater, 2024-05-06:
> Okay, I have solved this problem and the methods I have taken are as follows: (1) Firstly, search for "ISAAC SIM ASSETS PACK" in the Omnivers Launcher interface and download these three packages [...]

I met the same problem, and I fixed it by following these steps.

closed 2024-06-04 by Dhoeller19

destroy314, 2024-06-28:
I'm using Isaac sim 4.0.0 and change IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py line 599 into my local asset path.