isaac-sim/IsaacLab #399: [Question] Selecting objects to render in (or to filter out from) Camera sensors

titoirfan, 2024-05-06:
### Question

Hi, thanks for the great work on Orbit.

I am doing vision-based RL, so I have front-facing onboard cameras on my legged robots.

While the robots do not collide with each other during training, they still see each other in their cameras. This can clutter the robot's vision, especially if the number of parallel environments is high.

Is there a way to select which objects to (or not to) render in the robot's camera view?

Thanks for your time and support.

ninamortensen, 2024-05-23:
Hi there, do you mind sharing your setup? I can't seem to get the basic rendering to work (e.g. https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py). The demo only runs when the camera sensor is commented out, otherwise it throws the following error:

`Program aborted due to an unhandled Error:
Unable to find target for this triple (no targets are registered)Aborted (core dumped)`

titoirfan, 2024-05-24:
@ninamortensen I personally did not have any trouble running that demo script.

My setup is:
- Isaac Sim Version: 2023.1.1
- OS: Pop!_OS 22.04
- GPU: RTX 4060
- CUDA: 12.4
- GPU Driver: 550.67

I also have `libllvm11` ver. 1:11.1.0-6 and `libllvm15` ver. 1:15.0.7-0ubuntu0.22.04.3 if that helps. I followed the official tutorials for installing Isaac Sim and Orbit (using the Orbit conda environment).

The only trouble I had was when I tried to run the demo in headless mode, for which I figured out I have to use the `offscreen_rendering` flag.

ninamortensen, 2024-05-27:
> @ninamortensen I personally did not have any trouble running that demo script.
>
> My setup is:
>
>     * Isaac Sim Version: 2023.1.1
>
>     * OS: Pop!_OS 22.04
>
>     * GPU: RTX 4060
>
>     * CUDA: 12.4
>
>     * [...]

Got it, thanks a lot!

closed 2024-06-04 by Dhoeller19

hutslib, 2024-07-24:
> ### Question
> 
> Hi, thanks for the great work on Orbit.
> 
> I am doing vision-based locomotion, so I have front-facing onboard cameras on my robots.
> 
> While the robots do not collide with each other during training, they still see each other in their (RGB) cameras. This can clutter the robot's vision, especially if the number of parallel environments is high.
> 
> Is there a way to select which objects to (or not to) render in the robot's camera view?
> 
> Thanks for your time and support.

Hi, I meet the same problem and is this fixed?

DonovanZhu, 2025-01-16:
> ### Question
> Hi, thanks for the great work on Orbit.
>
> I am doing vision-based locomotion, so I have front-facing onboard cameras on my robots.
>
> While the robots do not collide with each other during [...]

I have a similar problem. To receive the depth image, I add a depth camera on the robot by using the following code:
```
depth_camera = TiledCameraCfg(
        prim_path="{ENV_REGEX_NS}/Robot/base_link/camera",
        height=480,
        width=848,
        data_types=["depth"],
        spawn=PinholeCameraCfg(
            focal_length=44.680277311912806, focus_distance=400.0, horizontal_aperture=84.8, clipping_range=(0.1, 1.5)
        ),
        offset=TiledCameraCfg.OffsetCfg(pos=(0.1, 0, 0), rot=(1.0, 0.0, 0.0, 0.0), convention="ros"),
    )
```
It do receive depth image, but other robots in the simulation also show up in the field of view. Is there any setting I can use so that other robots can be filtered, only terrain and current robot can be seen in the sensor?