isaac-sim/IsaacLab #406: 2 ModuleNotFoundError's when going through Orbit installation

mattmazzola, 2024-05-09:
I was attempting to go through the [Orbit Installation setup](https://isaac-orbit.github.io/orbit/source/setup/installation.html#building-extensions) and ran into some issues.

# Error 1: `ModuleNotFoundError: No module named 'toml'`

```
(orbit) ➜  orbit git:(main) ✗ ./orbit.sh --install
[INFO] Installing extensions inside orbit repository...                                                                                                                                                             
     module: /home/mattm/repos/orbit/source/extensions/omni.isaac.orbit_tasks
Obtaining file:///home/mattm/repos/orbit/source/extensions/omni.isaac.orbit_tasks
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/home/mattm/repos/orbit/source/extensions/omni.isaac.orbit_tasks/setup.py", line 10, in <module>
          import toml
      ModuleNotFoundError: No module named 'toml'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.
```

## Solution

I used the virtual environment setup and previously ran 
```
./orbit.sh --conda
```
I think this is supposed to use
```
./_isaac_sim/environment.yml
```
when creating the environment which has
```
name: isaac-sim
channels: 
  - defaults
  - pytorch
  - nvidia
dependencies: 
  - python=3.10
  - pip
  - pytorch 
  - torchvision 
  - torchaudio 
  - cuda-toolkit=11.7
  - pip:
    - stable-baselines3==2.0.0
    - tensorboard==2.11.0 
    - tensorboard-plugin-wit==1.8.1 
    - protobuf==3.20.3
```

I was able to fix by manually installing [`toml`](https://pypi.org/project/toml/)
```
pip install toml
```

# Error 2: `ModuleNotFoundError: No module named 'omni.isaac.kit'`

```
(orbit) ➜  orbit git:(main) ✗ python source/standalone/tutorials/00_sim/create_empty.py
Traceback (most recent call last):
  File "/home/mattm/repos/orbit/source/standalone/tutorials/00_sim/create_empty.py", line 20, in <module>
    from omni.isaac.orbit.app import AppLauncher
  File "/home/mattm/repos/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app/__init__.py", line 15, in <module>
    from .app_launcher import AppLauncher  # noqa: F401, F403
  File "/home/mattm/repos/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app/app_launcher.py", line 23, in <module>
    from omni.isaac.kit import SimulationApp
ModuleNotFoundError: No module named 'omni.isaac.kit'
```

## Solution

I am not sure how to solve this one.
I previously ran
```
./orbit.sh --install
```

I see `orbit.isaac.kit` in the `_isaac_sim/exts` folder so I assume it should be available

[image]

Any recommendations how to install the `orbit.isaac.kit` module?
(Or what step was supposed to implicitly install it?)

## System Info

Ubuntu 20.04.6 LTS
Isaac Sim 2023.1.1

mattmazzola, 2024-05-10:
From what I understand, the availability of the `orbit.isaac.kit` module is made possible by using the version of python in ISAACSIM_PATH

I tried to test this but I am not sure how. Here are details
```
(orbit) ➜  orbit git:(main) ✗ orbit -p  
[INFO] Using python from: /home/mattm/miniconda3/envs/orbit/bin/python                                                                                                                                              
Python 3.10.14 (main, May  6 2024, 19:42:50) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

```
(orbit) ➜  orbit git:(main) ✗ pip list           
Package                  Version     Editable project location
------------------------ ----------- -----------------------------------------------------------------
...
omni-isaac-orbit         0.16.2      /home/mattm/repos/orbit/source/extensions/omni.isaac.orbit
omni-isaac-orbit_assets  0.1.2       /home/mattm/repos/orbit/source/extensions/omni.isaac.orbit_assets
omni-isaac-orbit_tasks   0.6.1       /home/mattm/repos/orbit/source/extensions/omni.isaac.orbit_tasks
```

Confirms there not and `orbit.isaac.kit` avilable, but there are OTHER packages from building extensions

amirhosein-vedadi, 2024-05-10:
I have encountered a similar issue with the toml module when using the Python executable provided by Isaac Sim on Ubuntu 22.04. However, the problem was resolved when I installed Orbit using Conda environment.

mattmazzola, 2024-05-10:
I was able to get it working using the solution described here: https://github.com/NVIDIA-Omniverse/orbit/issues/103#issuecomment-1662112177

I think the full solution requires modifying `setup_conda_env.sh` and `setup_python.env.sh` to not have these bugs, but I will close for now so discussion can consolidate on the referenced issue

closed 2024-05-10 by mattmazzola