isaac-sim/IsaacLab #409: [Question] Franka robot asset model gravity and coriolis compensation

mikelasa, 2024-05-13:
Hello,

I'm working in a cartesian impedance control using a usd based franka model imported from: 

`from omni.isaac.orbit_assets import FRANKA_PANDA_HIGH_PD_CFG  # isort:skip`

This is how I load the robot:
```
# articulation
    Right_robot = FRANKA_PANDA_HIGH_PD_CFG.replace(prim_path="{ENV_REGEX_NS}/Right_robot")
    Right_robot.init_state = FRANKA_PANDA_HIGH_PD_CFG.init_state.replace(
        joint_pos={
            "panda_joint1": 0.0,
            "panda_joint2": -0.569,
            "panda_joint3": 0.0,
            "panda_joint4": -2.810,
            "panda_joint5": 0.0,
            "panda_joint6": 3.037,
            "panda_joint7": 0.741,
            "panda_finger_joint.*": 0.04,
        },
    )
    Right_robot.init_state.pos = (0.0, 0.0, 0.0)
    Right_robot.init_state.rot = (0.0, 0.0, 0.0, 0.0)
    Right_robot.spawn.rigid_props.disable_gravity = True
```
I'm able to activate a deactivate the gravity compensation term but I dont see anything related to the coriolis and centripetal term. Is integrated within the robot model or is something that I have to compute by myself?

My final torque equation demands a coriolis term that I need to compute.

Tanks in advance and regards,
Mikel

Mayankm96, 2024-07-04:
You can get these quantities from the root physx view methods.

Mayankm96, 2024-07-30:
Closing this issue since there's no activity and above solution should work.

closed 2024-07-30 by Mayankm96