isaac-sim/IsaacLab #414: [Question] Isaac sim 2023.1.1 stuck when running orbit existing scripts

zhengshuai1, 2024-05-15:
Hi,
I try to run orbit demo scripts with  isaac sim 2023.1.1;

The reference link is as follow:
https://isaac-orbit.github.io/orbit/source/setup/sample.html

For example, when I run:
`./orbit.sh -p source/standalone/demos/quadrupeds.py`

Then Isaac sim get stuck and the log is :
```
2024-05-15 06:00:53 [256,455ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp -- In </World/Origin3/Robot>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2023.1.1/Isaac/Samples/Orbit/Robots/ANYbotics/ANYmal-D/anymal_d.usd@ for reference introduced by @anon:0x113a18e0:World0.usd@</World/Origin3/Robot>. (recomposing stage on stage @anon:0x113a18e0:World0.usd@ <0x132b2030>)

2024-05-15 06:00:53 [256,467ms] [Warning] [omni.isaac.orbit.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/Origin3/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
2024-05-15 06:00:53 [256,467ms] [Warning] [omni.isaac.orbit.sim.utils] Could not perform 'modify_articulation_root_properties' on any prims under: '/World/Origin3/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
Traceback (most recent call last):
  File "/home/stardust/software/orbit/source/standalone/demos/quadrupeds.py", line 183, in <module>
    main()
  File "/home/stardust/software/orbit/source/standalone/demos/quadrupeds.py", line 171, in main
    scene_entities, scene_origins = design_scene()
  File "/home/stardust/software/orbit/source/standalone/demos/quadrupeds.py", line 91, in design_scene
    anymal_d = Articulation(ANYMAL_D_CFG.replace(prim_path="/World/Origin3/Robot"))
  File "/home/stardust/software/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 96, in __init__
    super().__init__(cfg)
  File "/home/stardust/software/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/rigid_object/rigid_object.py", line 58, in __init__
    super().__init__(cfg)
  File "/home/stardust/software/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py", line 75, in __init__
    self.cfg.spawn.func(
  File "/home/stardust/software/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/utils.py", line 270, in wrapper
    schemas.activate_contact_sensors(prim_paths[0], cfg.activate_contact_sensors)
  File "/home/stardust/software/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/schemas/schemas.py", line 511, in activate_contact_sensors
    raise ValueError(
ValueError: No contact sensors added to the prim: '/World/Origin3/Robot'. This means that no rigid bodies are present under this prim. Please check the prim path.
2024-05-15 06:00:53 [256,473ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/stardust/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-05-15 06:00:53 [256,475ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2024-05-15 06:00:53 [256,475ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-15 06:00:53 [256,476ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-05-15 06:00:53 [256,478ms] [Warning] [carb.audio.context] 1 contexts were leaked
2024-05-15 06:00:53 [256,499ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2024-05-15 06:00:53 [256,505ms] [Warning] [omni.core.ITypeFactory] Module /home/stardust/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```
I can download the .usd file using wget; Thus my network is normal;
http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2023.1.1/Isaac/Environments/Grid/default_environment.usd

How to solve this problem? Thanks!

My PC configuration
- ubuntu 22.04 - Isaac sim 2023.1.1 
- Nvidia driver-550.67
- NVIDIA RTX 4070S

swstbecrpgmail, 2024-06-27:
Hello, are you a developer in China? I have also met this problem. My mentor's suggestion is to download all the files you need and construct a local server.

closed 2024-10-04 by Dhoeller19

swstbecrpgmail, 2024-10-17:
You can also download all the isaacsim assets from omniverse launcher and change the load path of assets in the program.

shepera, 2024-11-30:
> You can also download all the isaacsim assets from omniverse launcher and change the load path of assets in the program.

Hello. I'm trying to download assets from nucleus. But it's too large and the link timeout after downloading about 10G resources. Do you have any suggestions for this? Thanks for help.

swstbecrpgmail, 2024-11-30:
> Hello. I'm trying to download assets from nucleus. But it's too large and the link timeout after downloading about 10G resources. Do you have any suggestions for this? Thanks for help.

you may download from this link: https://pan.quark.cn/s/65ff850489b4#/list/share

swstbecrpgmail, 2024-11-30:
> Hello. I'm trying to download assets from nucleus. But it's too large and the link timeout after downloading about 10G resources. Do you have any suggestions for this? Thanks for help.

You may download from this link: https://pan.quark.cn/s/65ff850489b4#/list/share

LittlePotatoChip, 2024-12-25:
> You may download from this link: https://pan.quark.cn/s/65ff850489b4#/list/share

It needs a quark vip.No,vip is not enough because there are more than 10000+ files. I bought a 3.8 yuan vip and find that it's not enough and then bought a 9.9 super vip(because I was first time to buy vip,it's about 30 yuan normally),and quark can't be used in linux. Now I'm trying to downloading it on my windows computer to a external hard drive. Good luck to me,hhhh

swstbecrpgmail, 2024-12-27:
> It needs a quark vip.No,vip is not enough because there are more than 10000+ files. I bought a 3.8 yuan vip and find that it's not enough and then bought a 9.9 super vip(because I was first time to [...]

Now，you may download the latest version of Isaaclab and isaacsim，the internet problem seems  to have been solved. I have never met the internet problems when using the latest version.