isaac-sim/IsaacLab #415: [Bug Report] no module named omni.usd

Guanbin-Huang, 2024-05-15:
[image]
i run the ./orbit.sh -p source/standalone/demos/quadrupeds.py

but i got an error. i googled it. i got , but as you can see below, the simulator launches firstly then everything else. i still don't know why it doens't work.
[image]

fyu-bdai, 2024-05-16:
Hi Guanbin, 

I had this issue yesterday as well. My issue was traced down to an incompatible isaac_sim version (2023.1.0 hotfix) with the current version of Orbit. The issue should be resolved if you're using isaac sim 2023.1.1.

cnalty, 2024-05-20:
I had this same issue. The documentation website still has the line in the installation guide

> export ISAACSIM_PATH="${HOME}/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1"

However you need to use isaac_sim-2023.1.1 . There is a warning at the top of the documentation version 2023.1.0 has been dropped, but this line has not been updated

closed 2024-10-04 by Dhoeller19