isaac-sim/IsaacLab #417: [Question] Run orbit tutorials scripts,but the system is error

lukewang-github, 2024-05-16:
Hello,
I try to run orbit tutorials scripts with isaac sim 2023.1.1;

The reference link is as follow:
https://isaac-orbit.github.io/orbit/source/tutorials/00_sim/spawn_prims.html

For example, when I run: ./orbit.sh -p source/standalone/tutorials/00_sim/spawn_prims.py

Then Isaac sim get stuck and the log is :

2024-05-16 13:32:10 [716,369ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp -- In </World/defaultGroundPlane>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2023.1.1/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:0x130016c0:World0.usd@</World/defaultGroundPlane>. (recomposing stage on stage @anon:0x130016c0:World0.usd@ <0x13acb3f0>)

Traceback (most recent call last):
  File "/home/luke/omniverse/share/ov/pkg/isaac_sim-2023.1.1/orbit20230515/source/standalone/tutorials/00_sim/spawn_prims.py", line 107, in <module>
    main()
  File "/home/luke/omniverse/share/ov/pkg/isaac_sim-2023.1.1/orbit20230515/source/standalone/tutorials/00_sim/spawn_prims.py", line 92, in main
    design_scene()
  File "/home/luke/omniverse/share/ov/pkg/isaac_sim-2023.1.1/orbit20230515/source/standalone/tutorials/00_sim/spawn_prims.py", line 44, in design_scene
    cfg_ground.func("/World/defaultGroundPlane", cfg_ground)
  File "/home/luke/omniverse/share/ov/pkg/isaac_sim-2023.1.1/orbit20230515/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "/home/luke/omniverse/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 628, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'
2024-05-16 13:32:10 [716,479ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/luke/omniverse/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-05-16 13:32:10 [716,483ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-05-16 13:32:10 [716,488ms] [Warning] [carb.audio.context] 1 contexts were leaked
2024-05-16 13:32:10 [716,511ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2024-05-16 13:32:10 [716,522ms] [Warning] [omni.core.ITypeFactory] Module /home/luke/omniverse/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.

How to solve this problem? Thanks!

My PC configuration

ubuntu 20.04 - Isaac sim 2023.1.1
Nvidia driver-535.86.05
NVIDIA RTX 4060

matheecs, 2024-05-24:
Make sure Nucleus is turned on. Offline [local assets pack](https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_faq.html#assets-pack) are recommended.

lukewang-github, 2024-05-25:
> 
Thank you for your reply. I resolved this issue using a local assets pack.

closed 2024-05-25 by lukewang-github

Subway2023, 2024-09-18:
> > 
> 
> Thank you for your reply. I resolved this issue using a local assets pack.

May i ask how to use a local assets pack. I face the same problem