isaac-sim/IsaacLab #424: [Bug Report] Crash lduring oading Nvidia Warehouse asset as terrain

jae1e, 2024-05-22:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I tried to load the `Simple_Warehouse/warehouse_multiple_shelves.usd` terrain using `TerrainImporterCfg`, and got crash.

### Steps to reproduce

My code looks like below

```
import omni.isaac.core.utils.nucleus as nucleus_utils

...

def _load_terrain(self, environment_asset):
        print("Loading terrain...")
        nvidia_assets_path = nucleus_utils.get_assets_root_path()
        environments_path = "/Isaac/Environments"
        usd_path = nvidia_assets_path + environments_path + "/" + environment_asset
        terrain_cfg = TerrainImporterCfg(
            num_envs=self.num_envs,
            prim_path=self.env_prim_path,
            terrain_type="usd",
            usd_path=usd_path,
            max_init_terrain_level=5,
            collision_group=-1,
            debug_vis=False,
        )
        terrain: TerrainImporter = terrain_cfg.class_type(terrain_cfg)
        print("Loaded terrain")

...

environment_asset="Simple_Warehouse/warehouse_multiple_shelves.usd"        self._load_terrain(environment_asset=environment_asset)
```

Then I got the error below:

```
Traceback (most recent call last):
  File "/home/jaeil/dev/OmniDrones/scripts/train_lidar.py", line 200, in main
    base_env = env_class(cfg, headless=cfg.headless)
  File "/home/jaeil/dev/rl/torchrl/envs/common.py", line 175, in __call__
    instance: EnvBase = super().__call__(*args, **kwargs)
  File "/home/jaeil/dev/OmniDrones/omni_drones/envs/warehouse.py", line 118, in __init__
    super().__init__(cfg, headless)
  File "/home/jaeil/dev/OmniDrones/omni_drones/envs/isaac_env.py", line 134, in __init__
    global_prim_paths = self._design_scene()
  File "/home/jaeil/dev/OmniDrones/omni_drones/envs/warehouse.py", line 184, in _design_scene
    self._load_terrain(environment_asset=environment_asset)
  File "/home/jaeil/dev/OmniDrones/omni_drones/envs/warehouse.py", line 211, in _load_terrain
    terrain: TerrainImporter = terrain_cfg.class_type(terrain_cfg)
  File "/home/jaeil/dev/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/terrains/terrain_importer.py", line 99, in __init__
    self.import_usd("terrain", self.cfg.usd_path)
  File "/home/jaeil/dev/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/terrains/terrain_importer.py", line 272, in import_usd
    faces = np.asarray(mesh_prim.GetFaceVertexIndicesAttr().Get()).reshape(-1, 3)
ValueError: cannot reshape array of size 4 into shape (3)
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Commit: 88cf9cda7ea77f206316ff65f2df6fac9aa6f7ca
- Isaac Sim Version: 2023.1.0-rc.42+2023.1.667.d641947f.tc(sim)
- OS: Ubuntu 22.04
- GPU: RTX 4070 Super
- CUDA: 12.2
- GPU Driver: 535.171.04

### Additional context

Add any other context about the problem here.

### Checklist

- [v] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Loading the Warehouse USD as terrain without crash

vmoens, 2024-05-22:
@btx0424 I have little context here but if this is something related to torchrl I'd be happy to help!

Ltesfaye, 2024-05-31:
The  TerrainImporterCfg doesn't currently support quads. If you read through the TerrainImporter implementation, it assumes that the terrain is coming in as triangles. If you're trying to load in the scene, you can do it using  sim_utils from omni.isaac.orbit.sim. This loader is able to launch USD models with quads and triangles but will not automatically support the physics collision filtering that is done as part of the terrain importer. 

```   
import omni.isaac.orbit.sim as sim_utils
...
def load_scenes(self):
        cfg = sim_utils.UsdFileCfg(usd_path=self.FULL_SCENE_MODEL_PATH)
        if self.ENABLE_COLLIDER:
            cfg.collision_props=sim_utils.CollisionPropertiesCfg(collision_enabled=True)

        model_orientation = (0.70711,0.70711,0.0,0.0) if self.LOADING_MODEL_WITH_Y_UP else None
        cfg.func("/World/Site/terrain", cfg,  orientation = model_orientation)
```

Mayankm96, 2025-03-25:
Should be fixed in: https://github.com/isaac-sim/IsaacLab/pull/1987

closed 2025-03-25 by Mayankm96