isaac-sim/IsaacLab #432: [Question] Writing Joint Velocity into Simulation with robot.write_joint_state_to_sim(position=..., velocity=..., joint_ids=...)

YZhang001, 2024-05-23:
### Question
Hello. I was simulating a Franka robot arm in Isaac Sim 2023.1.1 and followed the tutorial from [this page](https://isaac-orbit.github.io/orbit/source/tutorials/05_controllers/run_diff_ik.html). 

I need to assume perfect joint velocity tracking for the robot, so for every time step I tried to directly set the joint velocities with: 
`robot.write_joint_state_to_sim(position=..., velocity=..., joint_ids=robot_entity_cfg.joint_ids)`
But this function requires joint position as input, and if I set `position` as the current joint position and `velocity` as the desired joint velocity, the robot hardly moves in simulation. Is there a way to directly set joint velocity so that the robot can start moving from the current joint position with the set joint velocity during one control loop? 

Thanks.

jtigue-bdai, 2024-06-12:
Yes there is take a look at the [Articulation.set_joint_velocity_target](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.assets.html#omni.isaac.lab.assets.Articulation.set_joint_velocity_target), just be sure to read the section about writing that data to sim.

RandomOakForest, 2025-01-04:
Closing this question since it has been addressed and docs have been updated since. Thank you.