isaac-sim/IsaacLab #433: [Proposal] About including the Franka FR3 model in Orbit

YZhang001, 2024-05-24:
### Proposal

Hello, could you please include the Franka FR3 robot in orbit robot models? I read from the Isaac Sim 2023.1.1 release notes that it is now added. I'm asking because FR3 has different joint limits and maybe also other different specifications compared to the panda model. 

And also from `source/extensions/omni.isaac.orbit_assets/omni/isaac/orbit_assets/franka.py` it seems there is only proportional term for joint velocity tracking using `set_joint_velocity_target()` function (`stiffness` for joint position and `damping` for joint velocity so it could be a PD-controller for joint position but only a P-controller for joint velocity). Is it possible to somehow extend the controller to a PD controller for joint velocity? 

### Motivation

I have tried to add the FR3 model myself with orbit, using the USD path `usd_path=f"{ISAAC_NUCLEUS_DIR}/Robots/Franka/FR3/fr3.usd"` and following the code structure from `source/extensions/omni.isaac.orbit_assets/omni/isaac/orbit_assets/franka.py`. But I don't know how to specify the joint position limits for it, and when I run my script there is always warnings about the model: 
```
2024-05-24 13:24:50 [12,685ms] [Warning] [omni.physx.plugin] The rigid body at /World/envs/env_0/Robot/fr3_hand_tcp has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for.
2024-05-24 13:24:50 [12,685ms] [Warning] [omni.physx.plugin] The rigid body at /World/envs/env_0/Robot/fr3_link8 has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for.
```
I'm not sure whether this will affect the performance of joint velocity tracking and if there are other problems it might cause. If the FR3 model won't be included in orbit, what could I do to avoid these warnings? 

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)

Mayankm96, 2024-07-04:
While we'd like to add support to this model, we are currently swamped with many requests.

You can create the asset yourself using the steps here: https://isaac-sim.github.io/IsaacLab/source/how-to/import_new_asset.html

If you could import the asset and set up its configuration, we'd be happy to include it into Isaac Lab :)

RandomOakForest, 2025-01-04:
Thank you for submitting this. Let us know if need further help. Please post a new issue or discussion if you need so.

closed 2025-01-04 by RandomOakForest