isaac-sim/IsaacLab #435: Docker: Detected a blocking function. This will cause hitches or hangs in the UI. Please switch to the async version

swstbecrpgmail, 2024-05-27:
I run "./orbit.sh -p source/standalone/demos/quadrupeds.py --livestream 1" in docker, and it reports that
[image]
[image]
[image]

**System Info**
Ubuntu 20.04
Isaac Sim 2023.1.1
GPU: A6000
Driver Version: 550.67
CUDA Version: 12.4 

Additionally, I am a developer in China. I am not sure if this issue is related to network problems, although I have run the command multiple times.  And I have never met this issue before.

swstbecrpgmail, 2024-05-29:
I have confirmed that this is a network problem. If anyone has the similar problem or potential solutions, please leave a comment.

shenqidweisuo, 2024-12-25:
说是下载好资源，把他改成本地路径，但是细节你知道怎么做吗？
https://pan.quark.cn/s/65ff850489b4#/list/share/7e0542e0d62647a1aac4ef96629ebf51-isaac_sim%E5%90%88%E9%9B%86

RandomOakForest, 2025-01-04:
Since the old issue had been closed and it refers to a previous version of Isaac Sim, we would encourage you to try Isaac Sim 4.2 and Isaac Lab 1.4.0. We will close this issue for now. Please open a new one if you still face this problem. Thanks.

closed 2025-01-04 by RandomOakForest

swstbecrpgmail, 2025-01-11:
> 说是下载好资源，把他改成本地路径，但是细节你知道怎么做吗？ https://pan.quark.cn/s/65ff850489b4#/list/share/7e0542e0d62647a1aac4ef96629ebf51-isaac_sim%E5%90%88%E9%9B%86

[image]
You can modify the path as shown in the image, but it seems that in the latest version of IsaacLab, such issues no longer occur.

zhexulong, 2025-01-12:
> ![image](https://private-user-images.githubusercontent.com/153968462/402228133-d85c4602-63c4-4412-8391-a77f734952d2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

On my computer, this situation is still happening, and my Isaac lab is the latest.
sad.
But I am about to download the assets you provided on quark, wish me good luck