isaac-sim/IsaacLab #446: Resource paths don't exist.

Robokan, 2024-06-01:
### Describe the bug

Many of the examples don't work because the resources don't exist.

### Steps to reproduce
./isaaclab.sh -p source/standalone/demos/quadrupeds.py
or
./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Reach-Franka-v0 --headless
```
FileNotFoundError: USD file not found at path: 'http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2023.1.1/Isaac/IsaacLab/Robots/ANYbotics/ANYmal-B/anymal_b.usd'
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 2023.1.1
- OS: Ubuntu 20.04
- GPU: RTX 4090
- CUDA: 12.2
- GPU Driver: 535.54.03

### Additional context

Add any other context about the problem here.

### Checklist

- [x ] I have checked that there is no similar issue in the repo (**required**)
- [x ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Mayankm96, 2024-06-01:
Good catch. Can you replace this to: `f"{ISAAC_NUCLEUS_DIR}/Orbit"` for now? In Isaac Sim 2023.1.1, the assets are still available under "Orbit" directory on the Nucleus.

> https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/assets.py#L54-L55

JunghwanRo, 2024-06-05:
Were you able to run IsaacLab with 2023.1.1?

KevinGeng3, 2024-09-28:
same question

maxkra15, 2026-09-15:
Thanks @Robokan for reporting the missing asset paths.

This report targets Isaac Sim 2023.1.1 and its retired Nucleus/Orbit asset layout. Isaac Lab now supports Isaac Sim 6.1; Isaac Sim 5.1 and older are unsupported. We are closing this obsolete compatibility report. Please reopen with the exact URI and versions if a missing asset path still reproduces on a current Isaac Lab and Isaac Sim 6.1 installation.

closed 2026-09-15 by maxkra15