isaac-sim/IsaacLab #457: [Bug Report] Missing "Building your own project" template repository

AugustDG, 2024-06-04:
### Describe the bug

Before the Isaac Sim 4.0 update we had a repository to build our applications on top of, but it no longer exists.

### Steps to reproduce

Going to this doc [page](https://isaac-sim.github.io/IsaacLab/source/setup/template.html) and following the on-page links leads the user to a 404 Github page.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

joehays, 2024-06-05:
Agreed. Looking for: https://github.com/isaac-sim/IsaacLab.ext_template

arhanjain, 2024-06-14:
+1 on this

GRINGOLOCO7, 2024-06-21:
I still don't found the template

KyleM73, 2024-06-22:
you'll have to make some changes to make it compatible with current isaac lab (takes at most an afternoon of work) but the template is here:

https://github.com/isaac-orbit/orbit.ext_template

fishjohn, 2024-06-24:
I spent some time adapting the original [orbit.ext_template](https://github.com/isaac-orbit/orbit.ext_template) to make it compatible with IsaacLab. You can find the modified template here: [IsaacLab.ext_template](https://github.com/fishjohn/IsaacLab.ext_template). Please note that I cannot guarantee that it is completely error-free. Looking forward to the official version.

KyleM73, 2024-06-24:
nice! the only high level changes I see missing are in the top level `IsaacLab.ext_template/pyproject.toml` the `.gitignore` and `IsaacLab.ext_template/.vscode/tools/settings.template.json`. I also think the logic changed slightly in `IsaacLab.ext_template/.vscode/tools/setup_vscode.py`, I would look at the `.vscode` folder in Isaac Lab and copy that logic directly.

Mayankm96, 2024-06-25:
Sorry for the delay on this. We have now released the template: https://github.com/isaac-sim/IsaacLabExtensionTemplate

We are still testing it thoroughly, but feel free to try it out. If you face any problems, please don't hesitate to open an issue on the template repository.