isaac-sim/IsaacLab #463: Vscode intelligent prompt question

WSYANGSX, 2024-06-05:
Hi there!
   When I update Omniisaacgym to IsaacLab as office tutorials. I find it can't execute intelligent prompt in vscode about module relative with omni.isaac.lab... . How can i fix this. Thank you for help!
[image]

pxlong, 2024-06-06:
+1, i can not do the code navigation after runing **setup_python_env** from https://isaac-sim.github.io/IsaacLab/source/setup/developer.html

pxlong, 2024-06-06:
@WSYANGSX  hi, i think maybe the keypoint is to add the source/extensions/omni.isaac.lab* to the **python.analysis.extraPaths** in setttings.json in .vscode
[image]

WSYANGSX, 2024-06-07:
@pxlong Thank you for nice response! I will try it.

PR #526 (2024-06-25): Adds extensions to VSCode `python.analysis.extraPaths` settings
# Description

I added the path of isaac lab extensions to vscode python.analysis.extraPaths to execute intelligent prompt in vscode properly.

Fixes #463 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

In  .vscode/tools/setup_vscode.py
[image]

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documen
commit f486eb65 (2024-06-25): Adds extensions to VSCode `python.analysis.extraPaths` settings (#526)

# Description

This MR adds the path of the isaac lab extensions to vscode
`python.analysis.extraPaths` to execute intelligent prompt in vscode
properly.

Fixes #463 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

In  .vscode/tools/setup_vscode.py

[image]

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have [... cut]
commit 512710c7 (2024-06-25): Adds extensions to VSCode `python.analysis.extraPaths` settings (#526)

This MR adds the path of the isaac lab extensions to vscode
`python.analysis.extraPaths` to execute intelligent prompt in vscode
properly.

Fixes #463

- Bug fix (non-breaking change which fixes an issue)

In  .vscode/tools/setup_vscode.py

[image]

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my
feature wo [... cut]
commit ffec353e (2024-06-25): Adds IsaacLab extensions to VSCode settings (#526)

This MR adds the path of the isaac lab extensions to vscode
`python.analysis.extraPaths` to execute intelligent prompt in vscode
properly.

Fixes #463

- Bug fix (non-breaking change which fixes an issue)

In  .vscode/tools/setup_vscode.py

[image]

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my
feature works
- [x] I have run [... cut]
closed 2024-06-25 by Mayankm96