isaac-sim/IsaacLab #470: Possibility of switching control of robots for a bimanual teleoperation setup?

mikelasa, 2024-06-07:
Hello,

I used the code of the teleoperation example with the spacemouse to build my own environment where I have two robots. I'm interested in switching the control of both end effectors during the simulation, so I can move both robots for a manipulation task.

Is possible to do so? if so, I would appreciate any hint or orientation about how to develop this!

Thanks in advance

jtigue-bdai, 2024-06-07:
To clarify are you looking to be able to switch control between two arms using the same input or to directly control both arms at the same time with the same input?

mikelasa, 2024-06-07:
My idea is to switch the control of two arms with the same input as you say during the simulation.

But being honest the second option is also interesting.

jtigue-bdai, 2024-06-12:
You will have to create separate sets of actions for each robot and then handle the switching of applying the space mouse delta command somewhere before passing the the `action` to `step`.  Without knowing your whole code setup I would imagine this would happen in the `pre_process_actions`

mikelasa, 2024-06-12:
Understood! so far I dont have a manager based environment, I'm struggling with the implementation of it. But the idea is to develop a Bimanual manipulation environment with a custom Cartesian impedance controller to do imitation learning with the robots.