isaac-sim/IsaacLab #474: Setting num_channels=3 in CameraCfg in CartpoleCamera RGB task gives an error

dhruvkm2402, 2024-06-10:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

In the cartpole_camera task, the TiledCameraCfg works with 3 channels. However If I only want to use CameraCfg. It gives en error. But If I keep num_Channels for the CameraCfg as 4, it works.

### Steps to reproduce
Replace TiledCamera with 
# camera
    tiled_camera: CameraCfg = CameraCfg(
        prim_path="/World/envs/env_.*/Camera",
        offset=CameraCfg.OffsetCfg(pos=(-7.0, 0.0, 3.0), rot=(0.9945, 0.0, 0.1045, 0.0), convention="world"),
        data_types=["rgb"],
        spawn=sim_utils.PinholeCameraCfg(
            focal_length=24.0, focus_distance=400.0, horizontal_aperture=20.955, clipping_range=(0.1, 20.0)
        ),
        width=80,
        height=80,
    )
num_channels=3

Terminal Logs
'''
File "/home/dhruvm/IsaacLab/source/standalone/workflows/rl_games/train.py", line 150, in <module>
    main()
  File "/home/dhruvm/IsaacLab/source/standalone/workflows/rl_games/train.py", line 142, in main
    runner.run({"train": True, "play": False, "sigma": None})
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/torch_runner.py", line 133, in run
    self.run_train(args)
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/torch_runner.py", line 116, in run_train
    agent.train()
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 1318, in train
    step_time, play_time, update_time, sum_time, a_losses, c_losses, b_losses, entropies, kls, last_lr, lr_mul = self.train_epoch()
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 1182, in train_epoch
    batch_dict = self.play_steps()
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 742, in play_steps
    res_dict = self.get_action_values(self.obs)
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 408, in get_action_values
    res_dict = self.model(input_dict)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/algos_torch/models.py", line 262, in forward
    mu, logstd, value, states = self.a2c_network(input_dict)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/dhruvm/IsaacLab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/algos_torch/network_builder.py", line 399, in forward
    out = self.actor_cnn(out)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/container.py", line 217, in forward
    input = module(input)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/conv.py", line 460, in forward
    return self._conv_forward(input, self.weight, self.bias)
  File "/home/dhruvm/.local/share/ov/pkg/isaac-sim-4.0.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/nn/modules/conv.py", line 456, in _conv_forward
    return F.conv2d(input, weight, bias, self.stride,
RuntimeError: Given groups=1, weight of size [32, 3, 8, 8], expected input[2, 4, 80, 80] to have 3 channels, but got 4 channels instead

'''
Error
```
RuntimeError: Given groups=1, weight of size [32, 3, 8, 8], expected input[2, 4, 80, 80] to have 3 channels, but got 4 channels instead

```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.0.0
- OS: Ubuntu22.04
- GPU: RTX A6000
- CUDA: 12.0
- GPU Driver: 525.147.05

### Additional context

Add any other context about the problem here.

### Checklist

-  I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

Dhoeller19, 2024-06-10:
@dhruvkm2402 this example is meant to work for the tiled camera. If you want to use a normal camera, you have to modify it accordingly indeed.

closed 2024-06-10 by Dhoeller19

TTZYTttzyt, 2024-07-29:
But how to modify it accordingly indeed?Or I mean  I am not sure where to modify it to the proper channels?

commit d4e350e9 (2024-03-28): Fixes test scripts to report their status to the runner (#475)

# Description

@Mayankm96 recently found that tests failures weren't properly reporting
in runs of `orbit -t`. This is because we were previously relying on
individual test's main to exit with 1 if the test failed, but this
recently changed when we stopped them from exiting with `exit=False`
argument to `unittest.main()` to ensure the simulation app could be
closed properly.
 
We no longer call `simulation_app.close()` as this will cause the exit to not reach `run_all_tests.py`. The downside here is a bit more console spam.

Fixes [... cut]