isaac-sim/IsaacLab #479: self.sim.render() in direct_rl_env.py of IsaacLab takes too long (about 5 seconds)

LyuJZ, 2024-06-11:
Hi, I am using IsaacLab for object grasping based on RL. I used 4096 envs. In each env, there is a YCB object, a KUKA robot with a parallel gripper. The robot contact information is tracked by setting `activate_contact_sensors=True`. 

When I test the env, the warnings output is as flows:
```
2024-06-11 10:36:51 [20,564ms] [Warning] [omni.physx.plugin] The rigid body at /World/envs/env_0/Robot/root has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for.
2024-06-11 10:36:51 [20,564ms] [Warning] [omni.physx.plugin] The rigid body at /World/envs/env_0/Robot/tip has a possibly invalid inertia tensor of {1.0, 1.0, 1.0} and a negative mass, small sphere approximated inertia was used. Either specify correct values in the mass properties, or add collider(s) to any shape(s) that you wish to automatically compute mass properties for.
2024-06-11 10:38:02 [91,713ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /Visuals/ContactSensor.proto0_mesh_id0 that has not been populated
[INFO]: Time taken for simulation start : 95.219143 seconds
Creating window for environment.
[INFO]: Completed setting up the environment...
Time taken for rendering: 2.06 s
Time taken for rendering: 4.89 s
Time taken for rendering: 4.90 s
2024-06-11 10:38:43 [132,622ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 2190 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

2024-06-11 10:38:43 [132,623ms] [Warning] [omni.usd] Warning (secondary thread): in GetInstanceIndices at line 3256 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

2024-06-11 10:38:43 [132,625ms] [Warning] [omni.usd] Warning: in GetInstanceIndices at line 2190 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

2024-06-11 10:38:43 [132,625ms] [Warning] [omni.usd] Warning: in GetInstanceIndices at line 3256 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

Time taken for rendering: 4.81 s
2024-06-11 10:38:48 [137,575ms] [Warning] [omni.usd] Warning: in GetInstanceIndices at line 2190 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

2024-06-11 10:38:48 [137,575ms] [Warning] [omni.usd] Warning: in GetInstanceIndices at line 3256 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

Time taken for rendering: 4.83 s
2024-06-11 10:38:53 [142,840ms] [Warning] [omni.usd] Warning: in GetInstanceIndices at line 2190 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/pointInstancerAdapter.cpp -- ProtoIndex 1 out of bounds (prototypes size = 1) for (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

2024-06-11 10:38:53 [142,840ms] [Warning] [omni.usd] Warning: in GetInstanceIndices at line 3256 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usdImaging/usdImaging/delegate.cpp -- Empty InstanceIndices (/Visuals/ContactSensor, /Visuals/ContactSensor.proto1_mesh_id0)

```
The YCB object usd file I used is provided by NVIDIA, located in `YCB/Axis_Aligned/011_banana.usd`. The simulation time is much slower than OmniverseIsaacgymenvs I used before. Is this normal?

LyuJZ, 2024-06-11:
I think the [Warning] information is because I set the `debug_vis=True` in the `ContactSensorCfg`. After I set `debug_vis=False`, the [Warning] is gone. However, it still takes 2 seconds for every `self.sim.render()` step.

LyuJZ, 2024-06-11:
I guess I found the bug. I set the `Show By Type/Physics/Collidders` in the GUI to `All`, and this makes the rendering very slow. After I set it to `None`, the `self.sim.render()` just takes 0.02s for 4096 envs.

hojae-io, 2024-09-05:
Hi, I'm facing the similar issue. 
Can you elaborate what you mean by "I set the Show By Type/Physics/Collidders in the GUI to All" ?

It would really help! 
Thank you!