isaac-sim/IsaacLab #480: [Bug Report] Failed to install isaaclab via pip

dementrock, 2024-06-11:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Error: `No matching distribution found for isaacsim-robot-motion==4.0.0.0`

### Steps to reproduce

Followed the official doc to try to install isaac lab:

```bash
conda create -n isaaclab python=3.10
conda activate isaaclab
pip install torch==2.2.2 --index-url https://download.pytorch.org/whl/cu121
pip install isaacsim-rl isaacsim-replicator --index-url https://pypi.nvidia.com/
```

On the last step, getting

```
(isaaclab) ubuntu@ip-172-63-34-190:~$ pip install isaacsim-rl isaacsim-replicator --index-url https://pypi.nvidia.com/
Looking in indexes: https://pypi.nvidia.com/
Collecting isaacsim-rl
  Using cached https://pypi.nvidia.com/isaacsim-rl/isaacsim_rl-4.0.0.0-cp310-none-manylinux_2_34_x86_64.whl (611 kB)
Collecting isaacsim-replicator
  Using cached https://pypi.nvidia.com/isaacsim-replicator/isaacsim_replicator-4.0.0.0-cp310-none-manylinux_2_34_x86_64.whl (1.1 MB)
Collecting isaacsim-robot==4.0.0.0 (from isaacsim-rl)
  Using cached https://pypi.nvidia.com/isaacsim-robot/isaacsim_robot-4.0.0.0-cp310-none-manylinux_2_34_x86_64.whl (5.8 MB)
Collecting isaacsim-benchmark==4.0.0.0 (from isaacsim-rl)
  Using cached https://pypi.nvidia.com/isaacsim-benchmark/isaacsim_benchmark-4.0.0.0-cp310-none-manylinux_2_34_x86_64.whl (629 kB)
Requirement already satisfied: gymnasium in ./.local/lib/python3.10/site-packages (from isaacsim-rl) (0.29.0)
Collecting isaacsim-sensor==4.0.0.0 (from isaacsim-robot==4.0.0.0->isaacsim-rl)
  Downloading https://pypi.nvidia.com/isaacsim-sensor/isaacsim_sensor-4.0.0.0-cp310-none-manylinux_2_34_x86_64.whl (5.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.0/5.0 MB 59.7 MB/s eta 0:00:00
INFO: pip is looking at multiple versions of isaacsim-robot to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement isaacsim-robot-motion==4.0.0.0 (from isaacsim-robot) (from versions: none)
ERROR: No matching distribution found for isaacsim-robot-motion==4.0.0.0
```

### System Info

Describe the characteristic of your environment:

- Commit: N/A
- Isaac Sim Version: 4.0.0-rc.21+4.0.13872.3e3cb0c9.gl
- OS: Ubuntu 22.04.4 LTS
- GPU: NVIDIA A10G
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

This is running on EC2.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

erwincoumans, 2024-06-18:
The experimental pip packages were uploaded after this issue was created. Could you try it again?

noskill, 2024-06-27:
@dementrock I had the same issue with old glibc version. manylinux_2_34 means you need glibc 2.34

Mayankm96, 2024-07-04:
Closing this issue with a note added in #631

Basically, you can't do PIP installation on Ubuntu 20.04 as the default GLIBC there is 2.31 while the pip packages are only available for 2.34+ versions. If you face an issue with Ubuntu 22.04 and onwards, please don't hesitate to open an issue on Isaac Sim developers forum: https://forums.developer.nvidia.com/c/omniverse/simulation/69

closed 2024-07-04 by Mayankm96