isaac-sim/IsaacLab #487: [Bug Report] ModuleNotFoundError: No module named 'omni.kit.usd'

supersglzc, 2024-06-12:
```
Traceback (most recent call last):
  File "/home/supersglzc/code/IsaacLab/source/standalone/demos/quadrupeds.py", line 30, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/home/supersglzc/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 117, in __init__
    self._create_app()
  File "/home/supersglzc/code/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 536, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/supersglzc/Isaac/isaac-sim-2023.1.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 195, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "/home/supersglzc/Isaac/isaac-sim-2023.1.1/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
```

Hi, I have verified that my IsaacSim installation is correct. When I tried to run the verification for Isaac lab, it gives me this error. 

I saw some comments suggesting that it may due to the Nucleus, but I checked that my Nucleus is connected.

Dhoeller19, 2024-06-12:
Hi @supersglzc,
We recommend updating to IsaacSim 4.0.0.

If you want to keep using 2023.1.1, you have to use the backwards compatible experience file [here](https://github.com/isaac-sim/IsaacLab/blob/main/source/apps/isaaclab.backwards.compatible.kit). You can activate is by giving the path to that file as command line input with `--experience=...`

jbt-cmu, 2024-06-12:
How exactly do I update to this version of IsaacSim?

Thanks!

jtigue-bdai, 2024-06-12:
To install Isaac Sim 4.0.0 search for it in the Omniverse Launcher Exchange Apps. You will find two Isaac Sim apps that show up. Find the one that has 4.0.0 as the version. 4.0.0 will not show up as an option as an upgrade option the Isaac Sim 2023.1.1 version of the app that is already installed. 

There is also the experimental pip install method.

closed 2024-06-14 by Dhoeller19

sangtaeklee1, 2024-06-14:
> Hi @supersglzc, We recommend updating to IsaacSim 4.0.0.
> 
> If you want to keep using 2023.1.1, you have to use the backwards compatible experience file [here](https://github.com/isaac-sim/IsaacLab/blob/main/source/apps/isaaclab.backwards.compatible.kit). You can activate it by giving the path to that file as command line input with `--experience=...`

Hi, @Dhoeller19 Thanks for info.
Would you be able to explain this in more detail?
With which command should I pass command line input?
I've tried followings but it complains that `experience` is invalid argument.
```
./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py --experience=source/apps/isaaclab.backwards.compatible.kit
```

MasterYip, 2024-06-17:
> To install Isaac Sim 4.0.0 search for it in the Omniverse Launcher Exchange Apps. You will find two Isaac Sim apps that show up. Find the one that has 4.0.0 as the version. 4.0.0 will not show as an upgrade option in the Isaac Sim 2023.1.1 version of the app that is already installed.
> 
> There is also the experimental pip install method.

Hello, unfortunately I meet with this problem using the Isaac Sim 2023.1.1 which tagged with 4.0.0 version too. 
This is the version I have selected to install (from binary).
[image]
It passes installation verification but stucks here.
```bash
2024-06-17 06:09:59 [8,498ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[isaaclab.python-1.0.0] dependency: (name: 'omni.isaac.menu', tag: '', version: '', exact: 0, optional: 0) can't be satisfied. Available versions:
	 (none found)

2024-06-17 06:09:59 [8,498ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/source/standalone/tutorials/00_sim/create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 117, in __init__
    self._create_app()
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 536, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 195, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
```

And indeed, I cannot find the python package `omni.kit.usd` in isaacsim installation directory.
[image]

To solve this problem, I have tried to install omniverse kit and source its setup bash, however, it seems the version of `libgf.so` is incorrect.

The Kit version is 105.1.2:
[image]

```bash
source ~/.local/share/ov/pkg/kit-105.1.2/setup_python_env.sh
```

```bash
2024-06-17 06:27:46 [7,280ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[isaaclab.python-1.0.0] dependency: (name: 'omni.isaac.menu', tag: '', version: '', exact: 0, optional: 0) can't be satisfied. Available versions:
	 (none found)

2024-06-17 06:27:46 [7,280ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/source/standalone/tutorials/00_sim/create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 117, in __init__
    self._create_app()
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 536, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 195, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "/home/raymon/Documents/CodeSpace/IssacSim/IsaacLab/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 17, in <module>
    from omni.kit.usd import layers
  File "/home/raymon/.local/share/ov/pkg/kit-105.1.2/exts/omni.kit.usd.layers/omni/kit/usd/layers/__init__.py", line 1, in <module>
    from .impl import *
  File "/home/raymon/.local/share/ov/pkg/kit-105.1.2/exts/omni.kit.usd.layers/omni/kit/usd/layers/impl/__init__.py", line 1, in <module>
    from .extension import (
  File "/home/raymon/.local/share/ov/pkg/kit-105.1.2/exts/omni.kit.usd.layers/omni/kit/usd/layers/impl/extension.py", line 3, in <module>
    import omni.usd
  File "/home/raymon/.local/share/ov/pkg/kit-105.1.2/exts/omni.usd/omni/usd/__init__.py", line 1, in <module>
    from ._impl import *
  File "/home/raymon/.local/share/ov/pkg/kit-105.1.2/exts/omni.usd/omni/usd/_impl/__init__.py", line 2, in <module>
    from .._usd import *
ImportError: /home/raymon/.local/share/ov/pkg/kit-105.1.2/exts/omni.usd.libs/bin/libgf.so: undefined symbol: _ZN32pxrInternal_v0_22__pxrReserved__11TfMallocTag10_doTaggingE

```

MasterYip, 2024-06-17:
> To install Isaac Sim 4.0.0 search for it in the Omniverse Launcher Exchange Apps. You will find two Isaac Sim apps that show up. Find the one that has 4.0.0 as the version. 4.0.0 will not show as an [...]

Thanks for the solution! 
It is a little bit tricky to install the according version for it is folded in the `All releases builds` option
[image]

KevinGrey, 2024-08-20:
Hi there, I solve this bug by changing my code like this:
before:
```
from isaacsim import SimulationApp  
simulation_app = SimulationApp({"headless": False})
```
alter:
```
from omni.isaac.lab.app import AppLauncher
import argparse
parser = argparse.ArgumentParser(description="This script demonstrates different dexterous hands.")
AppLauncher.add_app_launcher_args(parser)
args_cli = parser.parse_args()
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app
```

Kirovsiki, 2024-11-11:
I'm sure the version of my pip is 4.0.0, but it also shows missing: ModuleNotFoundError: No module named 'omni.kit.usd'I feel very confused and look forward to your answer

chugunovyar, 2025-10-01:
`(isaaclab) ➜  IsaacLab git:(main) ✗ ./isaaclab.sh -p  ~/projects/my_second_tmp/my_second_tmp/scripts/rsl_rl/train.py
[INFO] Using python from: /home/crank/miniconda3/envs/isaaclab/bin/python                                                                                                                                                                                                         
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/crank/soft/IsaacLab/apps/isaaclab.python.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/crank/.local/lib/python3.11/site-packages/omni/data/Kit/Isaac-Sim/5.0/user.config.json'
[Info] [carb] Logging to file: /home/crank/.local/lib/python3.11/site-packages/omni/logs/Kit/Isaac-Sim/5.0/kit_20251001_211446.log
2025-10-01T18:14:46Z [2ms] [Warning] [omni.ext.plugin] [ext: extensions] Extensions config 'extension.toml' doesn't exist '/home/crank/soft/IsaacLab/source/extensions' or '/home/crank/soft/IsaacLab/source/extensions/config'
[0.015s] Failed to solve some dependencies locally, syncing with extension registry...
2025-10-01T18:14:46Z [5ms] [Warning] [omni.ext.plugin] [ext: extensions] Extensions config 'extension.toml' doesn't exist '/home/crank/soft/IsaacLab/source/extensions' or '/home/crank/soft/IsaacLab/source/extensions/config'
[0.015s] [ext: omni.kit.async_engine-0.0.3] startup
[0.017s] [ext: omni.client.lib-1.1.0] startup
[0.031s] [ext: omni.kit.registry.nucleus-0.0.0] startup
2025-10-01T18:14:49Z [3,603ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[isaaclab.python-2.2.1] dependency: 'isaacsim.asset.importer.urdf' = { version='^' } can't be satisfied. Available versions:
	 (none found)
 Platform incompatible packages:
	 - [isaacsim.asset.importer.urdf-2.3.14+106.5.0.lx64.d.cp310] (registry)
	 - [isaacsim.asset.importer.urdf-2.3.14+106.5.0.wx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.urdf-2.3.14+106.5.0.lx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.urdf-2.3.14+106.5.0.wx64.d.cp310] (registry)
	 - [isaacsim.asset.importer.urdf-2.3.11+106.5.0.lx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.urdf-2.3.11+106.5.0.wx64.d.cp310] (registry)
	 - [isaacsim.asset.importer.urdf-2.3.11+106.5.0.wx64.r.cp310] (registry)
	 - [isaacsim.asset.importer.urdf-2.3.11+106.5.0.lx64.d.cp310] (registry)
 Synced registries:
	 - kit/default         : found 797 packages 
	 - kit/sdk             : found 342 packages 
	 - kit/community       : found 459 packages 

2025-10-01T18:14:49Z [3,603ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/home/crank/projects/my_second_tmp/my_second_tmp/scripts/rsl_rl/train.py", line 48, in <module>
    app_launcher = AppLauncher(args_cli)
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/home/crank/soft/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 127, in __init__
    self._create_app()
  File "/home/crank/soft/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 811, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/crank/.local/lib/python3.11/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 245, in __init__
    from .utils import create_new_stage, open_stage
  File "/home/crank/.local/lib/python3.11/site-packages/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/utils.py", line 23, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
2025-10-01T18:14:49Z [3,605ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
`

Hi guys. Iv have been created template by command:
`./isaaclab.sh -n` 

I thould that app check every thing when create template ? Does it ? After run like described above and have no luck.