isaac-sim/IsaacLab #489: Random seeds and environment creation

melecmat, 2024-06-13:
Hi!

I found a potential bug. It seems to me that the random seeds does not have an impact on the environment randomization. In my examples I randomize the mass in a startup event (`mdp.add_body_mass`). 

I started printing the randomized weights and they are always different. Is this the desired behavior? I would expect the seeds to be respected here, especially when the documentation of the scripts claims: 'Seed used for the environment'.

hojae-io, 2024-08-29:
I observe the same issue. Even if the seed is deterministically set to 42, the resulting training curve (i.e. rewards etc..) are different.
Have you figured out how to resolve this issue?

hojae-io, 2024-08-30:
Maybe you wanna check: https://github.com/isaac-sim/IsaacLab/issues/904

Mayankm96, 2024-09-06:
Closing this issue as #904 raises the same concerns. The fix is under review: https://github.com/isaac-sim/IsaacLab/pull/940

closed 2024-09-06 by Mayankm96