isaac-sim/IsaacLab #516: [Bug Report] ModuleNotFoundError:No module named 'omni.isaac.kit‘ ; Had tried the issue #103 Solution

cidxb, 2024-06-14:
Hi, I am new to this repo
I have followed the instruction from https://isaac-sim.github.io/IsaacLab/source/setup/installation/verifying_installation.html to check my installation of IsaacLab.
With these two codes
```
# Option 1: Using the isaaclab.sh executable
# note: this works for both the bundled python and the virtual environment
./isaaclab.sh -p source/standalone/tutorials/00_sim/create_empty.py

# Option 2: Using python in your virtual environment
python source/standalone/tutorials/00_sim/create_empty.py 
```
And it returns
```
[INFO] Using python from: /home/paxini/anaconda3/envs/LLM/bin/python                     
Traceback (most recent call last):
  File "/home/xxx/IsaacLab/source/standalone/tutorials/00_sim/create_empty.py", line 20, in <module>
    from omni.isaac.lab.app import AppLauncher
  File "/home/xxx/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/__init__.py", line 15, in <module>
    from .app_launcher import AppLauncher  # noqa: F401, F403
  File "/home/xxx/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/app/app_launcher.py", line 27, in <module>
    from omni.isaac.kit import SimulationApp
ModuleNotFoundError: No module named 'omni.isaac.kit'
```

My Isaac Sim is 4.0.0 installed from the Omniverse. I have done a  `Create Isaac Sim Symbolic Link` according to the guide.

I have read the relative issue posts, i see there were plenty similar error occurred before the name of project change to IsaacLab
#269 
#406 
and #103 

I had tried the solution of #103, but it doesn't work. So I wonder if there are some changes causing the solution from #103  no longer apply-able.  
Please if someone counter the same issue and has solution for it, can share?

Dhoeller19, 2024-06-14:
Hi @cidxb,
Have you done `./isaaclab.sh -i` ?

cidxb, 2024-06-17:
> `./isaaclab.sh -i`

Yes, of course I have.
I have followed the instruction strictly.

adamphung-ho, 2024-06-17:
I got the same issue when verifying isaac lab installation. I switched my shell to zsh and followed solution in #103, but it didn't work.

fuyh20, 2024-06-17:
> > `./isaaclab.sh -i`
> 
> Yes, of course I have. I have followed the instruction strictly.

hello，maybe you can use bash and then execute `./isaaclab.sh -I`

cidxb, 2024-06-17:
> > > `./isaaclab.sh -i`
> > 
> > 
> > Yes, of course I have. I have followed the instruction strictly.
> 
> hello，maybe you can use bash and then execute `./isaaclab.sh -i`

Hi, I have tried your solution, still not working.

cidxb, 2024-06-17:
I personally lean on this is kind of path problem that python can't recognize the module? Since my Issac Sim is running well... It is and I can import the sub modules of omni.isaac . I just don't really know how to fix it ...

adamphung-ho, 2024-06-18:
@cidxb you can try to run verifying code with bash shell in vscode. I tried and it worked.

cidxb, 2024-06-18:
> @cidxb you can try to run verifying code with bash shell in vscode. I tried and it worked.

I tried, i am using Ubuntu20.04

adamphung-ho, 2024-06-18:
Yeah, you can choose different type of shell with vscode, but the default is bash.

sunmingang, 2024-07-02:
I have the same issue, Does anyone have a solution  ? 
env: ubuntu22.04,  isaacsim 4.0,  installed by binary.

sunmingang, 2024-07-02:
The issue is solved, if you running the app/command at conda env, you should run the script "source setup_conda_env.sh", the file located the  Isaacsim root directory.

swstbecrpgmail, 2024-07-03:
> The issue is solved, if you running the app/command at conda env, you should run the script "source setup_conda_env.sh", the file located the Isaacsim root directory.

This works for me, thank you very much!

swstbecrpgmail, 2024-07-03:
This problem is probably caused by conda
Besides the "ModuleNotFoundError: No module named 'omni.isaac.kit'" problem, I also met the bash problem when running “./isaaclab.sh --conda”, as follows:
[image]
(The Chinese in the picture means "no such file or directory".)

Then,I modified the "source" part of the conda's "set_env.sh" for this environment as follows: (my username is crp)
[image]

And if you install isaac sim using omniverse launcher, there should exists a directory "_isaac_sim" in your home directory by default.
[image]

RunpeiDong, 2024-07-04:
Hi @sunmingang,

It does not work for me. What exact command do you run?

> The issue is solved, if you running the app/command at conda env, you should run the script "source setup_conda_env.sh", the file located the Isaacsim root directory.

swstbecrpgmail, 2024-07-05:
> Hi @sunmingang,
> 
> It does not work for me. What exact command do you run?
> 
> > The issue is solved, if you running the app/command at conda env, you should run the script "source setup_conda_env.sh", the file located the Isaacsim root directory.
Have you run this correctly?
[image]

RunpeiDong, 2024-07-05:
> Hi @sunmingang,
> It does not work for me. What exact command do you run?
> The issue is solved, if you running the app/command at conda env, you should run the script "source setup_conda_env.sh", the [...]

Hi @swstbecrpgmail,

Sure, I have done this. And I just ran the source setup_conda_env.sh from the _isaac_sim folder. But it just does not work.

RunpeiDong, 2024-07-05:
> Hi @swstbecrpgmail,
>
> Sure, I have done this. And I just ran the source setup_conda_env.sh from the _isaac_sim folder. But it just does not work.

Hi @swstbecrpgmail and @sunmingang,

After I first switched to `bash` and then run `source`, it finally worked... I think this should be fixed for the Zsh and Conda users. This is very inconvenient to use and may be confusing.

RunpeiDong, 2024-07-05:
One simple method to address this if you are using zsh and conda. Write a script `isaaclabpy.sh`:

```bash
#!/usr/bin/env bash

# Exits if error occurs
set -e

# Set tab-spaces
tabs 4

cd _isaac_sim
source setup_conda_env.sh
cd ..
echo "$1"
python $1
```

And set it as executable file: 
```
sudo chmod u+x isaaclabpy.sh
```
Then use `./isaaclabpy.sh` instead of `./isaaclab.sh` when you only run Python scripts.

closed 2024-07-05 by Mayankm96

Mayankm96, 2024-07-05:
Regarding ZSH support, we are looking into it.

RunpeiDong, 2024-07-05:
> Regarding ZSH support, we are looking into it.

Great! Looking forward to it.

Logan1904, 2024-07-12:
I ran in to the same problem too. 

My workflow: Followed the guide https://isaac-sim.github.io/IsaacLab/source/setup/installation/index.html, following the options for Windows and binary installation of IsaacSim

Problem: At the verification stage, I encountered this error: ModuleNotFoundError:No module named 'omni.isaac.kit‘

Solution: Follow this guide: https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_python.html. The missing Python modules will be installed with the 2 following commands:
`pip install isaacsim==4.0.0.0 --extra-index-url https://pypi.nvidia.com`
`pip install isaacsim-extscache-physics==4.0.0.0 isaacsim-extscache-kit==4.0.0.0 isaacsim-extscache-kit-sdk==4.0.0.0 --extra-index-url https://pypi.nvidia.com
`

Perhaps these steps should be added to the IsaacLab guide as well?

RunpeiDong, 2024-07-12:
> I ran in to the same problem too.
>
> My workflow: Followed the guide https://isaac-sim.github.io/IsaacLab/source/setup/installation/index.html, following the options for Windows and binary [...]

The solution using pip to install has been added to the guideline as an alternative method to set up. However, this method does not typically work as it has some strict requirements such as system version. In my case, I cannot install isaacsim with pip. This pip installation seems still under development, and I suppose most people would download isaacsim using launcher instead.

Logan1904, 2024-07-13:
Yep, I agree. What I am saying is that when downloading isaacsim using the omniverse launcher, it doesnt provide the required Python modules. The only way to get the omni.isaac.kit modules is to follow this guide for Python setup: https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_python.html. I think it should be made clearer in the IsaacLab guide

nicolasdoan, 2024-09-13:
can you tell which specific version of omni.isaac.kit did you managed to download from this command? "pip install isaacsim-omni.isaac.kit==4.0.0.0 --extra-index-url https://pypi.nvidia.com"?

krishnapraveen7, 2024-09-13:
Fkn re tarded guide with useless setup steps. Why add step with binaries when it doesn't work? If I have to install with pip install isaacsim then only write that , god how is it so fkn hard. So i have spent 2 hours to install useless isaac sim from their bullshit omniverse launcher just to know that i have to do pip install.

krishnapraveen7, 2024-09-13:
> Fkn re tarded guide with useless setup steps. Why add step with binaries when it doesn't work? If I have to install with pip install isaacsim then only write that , god how is it so fkn hard. So i [...]

Yes i get this bullshit error. I followed all the steps that are mentioned in the useless guide.

Traceback (most recent call last):
  File "C:\Users\krishna\ProjectX\IsaacLab\source\standalone\tutorials\00_sim\create_empty.py", line 20, in <module>
    from omni.isaac.lab.app import AppLauncher
  File "C:\Users\krishna\ProjectX\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\__init__.py", line 15, in <module>
    from .app_launcher import AppLauncher  # noqa: F401, F403
  File "C:\Users\krishna\ProjectX\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 26, in <module>
    from omni.isaac.kit import SimulationApp
ModuleNotFoundError: No module named 'omni.isaac.kit'

Subway2023, 2024-09-17:
> The issue is solved, if you running the app/command at conda env, you should run the script "source setup_conda_env.sh", the file located the Isaacsim root directory.

May I ask where the file setup_conda_env.sh is located?

zlw21gxy, 2024-10-11:
I switched from zsh to bash, and it has solved my problem. I am using Ubuntu 20.04.

Txxxxxxxxx, 2024-11-25:
I think I found a solution. You need to modify this file: **setup_python_env.sh** , which is referenced by **setup_conda_env_zsh.sh** . 

**setup_python_env.sh** :
`#!/bin/bash

if [ -n "$ZSH_VERSION" ]; then
    SCRIPT_DIR="$( cd "$( dirname "$0" )" && pwd )"
else
    SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
fi

export PYTHONPATH=$PYTHONPATH:$SCRIPT_DIR/../../../$PYTHONPATH:$SCRIPT_DIR/kit/python/lib/python3.10/site-packages:$SCRIPT_DIR/python_packages:$SCRIPT_DIR/exts/omni.isaac.kit:$SCRIPT_DIR/kit/kernel/py:$SCRIPT_DIR/kit/plugins/bindings-python:$SCRIPT_DIR/exts/omni.isaac.lula/pip_prebundle:$SCRIPT_DIR/exts/omni.exporter.urdf/pip_prebundle:$SCRIPT_DIR/extscache/omni.kit.pip_archive-0.0.0+10a4b5c0.lx64.cp310/pip_prebundle:$SCRIPT_DIR/exts/omni.isaac.core_archive/pip_prebundle:$SCRIPT_DIR/exts/omni.isaac.ml_archive/pip_prebundle:$SCRIPT_DIR/exts/omni.pip.compute/pip_prebundle:$SCRIPT_DIR/exts/omni.pip.cloud/pip_prebundle
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$SCRIPT_DIR/../../../$LD_LIBRARY_PATH:$SCRIPT_DIR/.:$SCRIPT_DIR/exts/omni.usd.schema.isaac/plugins/IsaacSensorSchema/lib:$SCRIPT_DIR/exts/omni.usd.schema.isaac/plugins/RangeSensorSchema/lib:$SCRIPT_DIR/exts/omni.isaac.lula/pip_prebundle:$SCRIPT_DIR/exts/omni.exporter.urdf/pip_prebundle:$SCRIPT_DIR/kit:$SCRIPT_DIR/kit/kernel/plugins:$SCRIPT_DIR/kit/libs/iray:$SCRIPT_DIR/kit/plugins:$SCRIPT_DIR/kit/plugins/bindings-python:$SCRIPT_DIR/kit/plugins/carb_gfx:$SCRIPT_DIR/kit/plugins/rtx:$SCRIPT_DIR/kit/plugins/gpu.foundation`

jiangranlv, 2024-12-26:
> I think I found a solution. You need to modify this file: **setup_python_env.sh** , which is referenced by **setup_conda_env_zsh.sh** .
>
> **setup_python_env.sh** : `#!/bin/bash
>
> if [ -n [...]

It works well. Thanks

iui-whgi, 2025-01-28:
시발 엔비디아

zhwhong, 2025-08-31:
> I think I found a solution. You need to modify this file: **setup_python_env.sh** , which is referenced by **setup_conda_env_zsh.sh** .
>
> **setup_python_env.sh** : `#!/bin/bash
>
> if [ -n [...]

It really helps. Thanks.

PR #631 (2024-07-05): Fixes `isaaclab` scripts to deal with Isaac Sim pip installation
# Description

Earlier the script hard-coded with path to Isaac Sim installation as the sym-linked directory "_isaac_sim". However, when doing PIP installation, this workflow doesn't hold. The MR modifies the scripts `isaaclab.sh` and `isaaclab.bat` to now check for PIP package installation and get paths from there if they exist.

Fixes #516, #629

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format` [... cut]
commit 07f082f5 (2024-07-05): Fixes `isaaclab` scripts to deal with Isaac Sim pip installation (#631)

# Description

Earlier the script hard-coded with path to Isaac Sim installation as the
sym-linked directory "_isaac_sim". However, when doing PIP installation,
this workflow doesn't hold. The MR modifies the scripts `isaaclab.sh`
and `isaaclab.bat` to now check for PIP package installation and get
paths from there if they exist.

Fixes #516, #629

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh - [... cut]