isaac-sim/IsaacLab #518: [Bug Report] Only during training, encounter "PhysX error: PxRigidBody::setMass(): mass must be non-negative!" despite non-negative/non-zero masses

Easternspork, 2024-06-15:
### Describe the bug

During training only, I encounter a `PhysX error: PxRigidBody::setMass(): mass must be non-negative!` error despite having non-negative/non-zero masses. The training does start, however, but the mean episode length is 1, so no there is no training being done, and thus I am assuming that this is the culprit. I have tried to isolate the issue by creating a seperate scene with all of the same configs as the training environment and running it, and I have no issues running there.  

Please let me know if this is a Isaac Sim bug instead, so I can ask in their forums - however because it only occurs during training, I am suspecting it is an issue with Isaac Sim. 

### Steps to reproduce

Because of such a vague error message, I can't really trace it to anything, so I'm having a hard time coming up with a minimal example, so I have listed the output, USD files, as well as some code snippets that are maybe relevant to the issue. Please let me know if there are other pieces of code/outputs I should submit.

[terminal_out.txt](https://github.com/user-attachments/files/15845921/terminal_out.txt)
[mistletoe_usd.zip](https://github.com/user-attachments/files/15845950/mistletoe_usd.zip)

Articulation Config

```py
HAA_ACTUATOR_CFG = DCMotorCfg(
    joint_names_expr=[".*HAA"],
    saturation_effort=24,
    effort_limit=24,
    velocity_limit=7.5,
    stiffness={".*": 40.0},
    damping={".*": 5.0},
)

KFE_ACTUATOR_CFG = DCMotorCfg(
    joint_names_expr=[".*KFE"],
    saturation_effort=9,
    effort_limit=9,
    velocity_limit=7.5,
    stiffness={".*": 40.0},
    damping={".*": 5.0},
)

HFE_ACTUATOR_CFG = DCMotorCfg(
    joint_names_expr=[".*HFE"],
    saturation_effort=9,
    effort_limit=9,
    velocity_limit=7.5,
    stiffness={".*": 40.0},
    damping={".*": 5.0},
)

#Articulation Configs

MISTLETOE_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path= dirname + '/assets/mistletoe_usd/mistletoe.usd',
        
        # Adapted from Anymal Velocity Tracking Example's Config
        activate_contact_sensors=True,
        
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            retain_accelerations=False,
            linear_damping=0.0,
            angular_damping=0.0,
            max_linear_velocity=1000.0,
            max_angular_velocity=1000.0,
            max_depenetration_velocity=1.0,
        ),

        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=True, solver_position_iteration_count=4, solver_velocity_iteration_count=0
        ),

    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.15),
        joint_pos={
            ".*HAA": 0.0,  # all HAA
            "LEG_[14]_HFE": 0.8,  # both front HFE
            "LEG_[23]_HFE": -0.8,  # both hind HFE
            "LEG_[23]_KFE": -0.8,  # both front KFE
            "LEG_[14]_KFE": 0.8,  # both hind KFE
        },
    ),

    actuators={
        "HAA": HAA_ACTUATOR_CFG,
        "HFE": HFE_ACTUATOR_CFG,
        "KFE": KFE_ACTUATOR_CFG
    },

    soft_joint_pos_limit_factor=0.95
)
```

Environment Config

```py
@configclass
class MistletoeRoughEnvCfg(LocomotionVelocityRoughEnvCfg):
    def __post_init__(self):
        # post init of parent
        super().__post_init__()
        # switch robot to anymal-c
        self.scene.robot = MISTLETOE_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")

@configclass
class MistletoeRoughEnvCfg_PLAY(MistletoeRoughEnvCfg):
    def __post_init__(self):
        # post init of parent
        super().__post_init__()

        # make a smaller scene for play
        self.scene.num_envs = 50
        self.scene.env_spacing = 2.5
        # spawn the robot randomly in the grid (instead of their terrain levels)
        self.scene.terrain.max_init_terrain_level = None
        # reduce the number of terrains to save memory
        if self.scene.terrain.terrain_generator is not None:
            self.scene.terrain.terrain_generator.num_rows = 5
            self.scene.terrain.terrain_generator.num_cols = 5
            self.scene.terrain.terrain_generator.curriculum = False

        # disable randomization for play
        self.observations.policy.enable_corruption = False
        # remove random pushing event
        self.events.base_external_force_torque = None
        self.events.push_robot = None
```

### System Info

Describe the characteristic of your environment:

- Commit: b468546
- Isaac Sim Version: 4.0.0
- OS: Ubuntu 22.04.3
- GPU: RTX 3070
- CUDA: 11.5
- GPU Driver: 535.171.04

### Additional context

The code I wrote is very similar to that of Anymal's environment, but running Anymal's environment seems to work, but mine doesn't. 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Easternspork, 2024-06-16:
Stupid mistake - the mass distribution parameter for randomizing rigid body mass was too large, causing the mass to be negative.

closed 2024-06-16 by Easternspork

zma69650, 2024-08-21:
@Easternspork  Hello, I encountered this issue while training the Unitree B1 as well. Could you please tell me where to adjust the 'random mass' parameter?

Easternspork, 2024-09-28:
Oops, sorry didn't realize this until now. Hope you figured it out, but just in case, the parameter would be set in your EventCfg on startup. ie. 
`@configclass
class EventCfg:
    """Configuration for events."""

    # startup
    add_base_mass = EventTerm(
        func=mdp.randomize_rigid_body_mass,
        mode="startup",
        params={
            "asset_cfg": SceneEntityCfg("robot", body_names="base"),
            "mass_distribution_params": (-5.0, 5.0), # << this param here
            "operation": "add",
        },
    )`