isaac-sim/IsaacLab #520: Issues getting observations for two robots with mdp.joint_pos_rel and mdp.joint_vel_rel

mikelasa, 2024-06-16:
Hello,

I'm working with two franka robots in a simulation where I'm trying to configure the observations for bot of them. My robots are called robot1 and robot2:

```
# robot 1
    robot1: ArticulationCfg = FRANKA_PANDA_HIGH_PD_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot1")
    robot1.actuators["panda_shoulder"].stiffness = 0.0
    robot1.actuators["panda_shoulder"].damping = 0.0
    robot1.actuators["panda_forearm"].stiffness = 0.0
    robot1.actuators["panda_forearm"].damping = 0.0
    robot1.init_state.pos = (0.0, 0.5, 0.0)
    
    # robot 2
    robot2: ArticulationCfg = FRANKA_PANDA_HIGH_PD_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot2")
    robot2.actuators["panda_shoulder"].stiffness = 0.0
    robot2.actuators["panda_shoulder"].damping = 0.0
    robot2.actuators["panda_forearm"].stiffness = 0.0
    robot2.actuators["panda_forearm"].damping = 0.0
    robot2.init_state.pos = (0.0, 1.0, 0.0)
```

My issue is that when I try to get the observations for position for example, I get an error telling that the asset "robot" doesn't exist: 
Traceback (most recent call last):
```
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/standalone/cartesian_impedance_control/env/bimanual_robot_teleop.py", line 179, in <module>
    main()
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/standalone/cartesian_impedance_control/env/bimanual_robot_teleop.py", line 39, in main
    env = ManagerBasedEnv(cfg=env_cfg)
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 117, in __init__
    self.load_managers()
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 205, in load_managers
    self.observation_manager = ObservationManager(self.cfg.observations, self)
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 41, in __init__
    super().__init__(cfg, env)
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/manager_base.py", line 130, in __init__
    self._prepare_terms()
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/observation_manager.py", line 254, in _prepare_terms
    obs_dims = tuple(term_cfg.func(self._env, **term_cfg.params).shape[1:])
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/observations.py", line 117, in joint_pos_rel
    asset: Articulation = env.scene[asset_cfg.name]
  File "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 352, in __getitem__
    raise KeyError(f"Scene entity with key '{key}' not found. Available Entities: '{all_keys}'")
KeyError: "Scene entity with key 'robot' not found. Available Entities: '['terrain', 'robot1', 'robot2', 'cube', 'ground', 'dome_light', 'table']'"
```
This is my ObservatioinsCfg: 
```
@configclass
class ObservationsCfg:

    @configclass
    class LowLevelActionsCfg(ObsGroup):
        # robot 1 joint position and velocity
        # mdp needs robot asset name to be robot1
        joint_pos1 = ObsTerm(func=mdp.joint_pos_rel)
        joint_vel1 = ObsTerm(func=mdp.joint_vel_rel)

        def __post_init__(self):
            self.enable_corruption = False
            self.concatenate_terms = False

    # observation groups
    LowLevelActions: LowLevelActionsCfg = LowLevelActionsCfg()
```
Seems that the `mdp.joint_pos_rel and mdp.joint_vel_rel` uses as default argument "robot" for the asset and I don't know how can I avoid this issue. 

```
def joint_pos_rel(env: ManagerBasedEnv, asset_cfg: SceneEntityCfg = SceneEntityCfg("robot")) -> torch.Tensor:
    """The joint positions of the asset w.r.t. the default joint positions.

    Note: Only the joints configured in :attr:`asset_cfg.joint_ids` will have their positions returned.
    """
    # extract the used quantities (to enable type-hinting)
    asset: Articulation = env.scene[asset_cfg.name]
    return asset.data.joint_pos[:, asset_cfg.joint_ids] - asset.data.default_joint_pos[:, asset_cfg.joint_ids]
```

In order to manage two robot observations, which is the way to get both positions and velocities?
Thanks in advance!

mikelasa, 2024-06-16:
I solved the issue, I didnt knew how to pass my own robot name, writing: 

joint_pos1 = ObsTerm(func=mdp.joint_pos_rel, params={"asset_cfg": SceneEntityCfg("robot1")})

solved my problem

closed 2024-06-16 by mikelasa

Hokite, 2024-08-13:
Can I share your Franka code? I want to refer to your code and use it for my research.

mikelasa, 2024-08-14:
Hi @Hokite, no worries and feel free to use it!

Hokite, 2024-08-27:
Thanks @mikelasa. I would like to ask additional questions, is it possible? I'm looking to learn two agents, what reinforcement learning packages did you use (rl-games, skrl, etc...)?

mikelasa, 2024-09-02:
Hi Hokite, I didnt used yet any of those packages. I'm actually building my own environment so I can the use them, so sorry, I cant help you here...