isaac-sim/IsaacLab #530: Imported urdf loads only the stage tree but not the robot

mikelasa, 2024-06-18:
Hello,

I'm unable to load my own panda model, which is a copy of the:

jtigue-bdai, 2024-06-18:
For your robot2, it may be having issues with references to the mesh assets in : "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab_assets/data/Robots/franka/panda.usd".

Is that usd an imported URDF? If so you can do a direct import in the scene config using the UrdfFileCfg instead of the UsDFileCfg.

jbt-cmu, 2024-06-23:
> For your robot2, it may be having issues with references to the mesh assets in : "/home/mikel/IsaacOrbitLFD/IsaacLab/source/extensions/omni.isaac.lab_assets/data/Robots/franka/panda.usd".
>
> Is that [...]

I have noticed that if you load any URDF with Isaac Lab, the RGB information is terribly wrong (its typically all white). The same happens with USDView in Omniverse, even when loading Nvidia's Carter and exporting to USD.

jtigue-bdai, 2024-06-24:
How is your URDF specifying RGB information? Are you utilizing .dae (collada files) or the <material><color rgba> route?

jbt-cmu, 2024-06-24:
My URDF, the Hello Robot stretch, does it like so:

```
<robot name="stretch">
  <link name="base_link">
    <inertial>
      <origin rpy="0 0 0" xyz="-0.087526 -0.001626 0.081009"/>
      <mass value="17.384389"/>
      <inertia ixx="0.160002" ixy="0.006758" ixz="0.004621" iyy="0.138068" iyz="0.002208" izz="0.228992"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="./meshes/base_link.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.79216 0.81961 0.93333 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="./meshes/base_link_collision.STL"/>
      </geometry>
    </collision>
  </link>
```

Mayankm96, 2024-07-02:
@mikelasa All zeros is an invalid quaternion. Can you fix this and see if that solves your problem?

```python
    robot2.init_state.rot = (0.0, 0.0, 0.0, 0.0)
```

@jbt-cmu Regarding textures, let's discuss that in the issues you raised. It's very likely that you need to post this issue on the Isaac Sim forums. The issue seems to be with URDF importer from Isaac Sim and not Isaac Lab.

mikelasa, 2024-07-03:
@Mayankm96 Yes, I already commented above that this was the issue, at some point I changed that to 0 the robot wasn't spawning, thank you

closed 2024-07-03 by mikelasa