isaac-sim/IsaacLab #541: [Bug Report] UR10 reach environment won't accept all wrappers

eferreirafilho, 2024-06-19:
### Describe the bug

UR10 reach can be trained with RSL_RL but not with RL_GAMES and SKRL.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

SKRL
```
./isaaclab.sh -p source/standalone/workflows/skrl/train.py --task=Isaac-Reach-UR10-v0 --max_iterations=10000 --num_envs=1024 --headless
```

or 

RL_GAMES
```
./isaaclab.sh -p source/standalone/workflows/rl_games/train.py --task=Isaac-Reach-UR10-v0 --max_iterations=10000 --num_envs=1024 --headless
```

Both gives errors like this:

```
[INFO]: Parsing configuration from: <class 'omni.isaac.lab_tasks.manager_based.manipulation.reach.config.ur_10.joint_pos_env_cfg.UR10ReachEnvCfg'>
Traceback (most recent call last):
  File "/home/edson/omniverse-playground/IsaacLab/source/standalone/workflows/skrl/train.py", line 184, in <module>
    main()
  File "/home/edson/omniverse-playground/IsaacLab/source/standalone/workflows/skrl/train.py", line 70, in main
    experiment_cfg = load_cfg_from_registry(args_cli.task, "skrl_cfg_entry_point")
  File "/home/edson/omniverse-playground/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 70, in load_cfg_from_registry
    mod_path = os.path.dirname(importlib.import_module(mod_name).__file__)
  File "/home/edson/.local/share/ov/pkg/isaac-sim-4.0.0/kit/python/lib/python3.10/posixpath.py", line 152, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

-->

### System Info

Describe the characteristic of your environment:

- Commit: 9ab6b48
- Isaac Sim Version: 4.0.0
- OS: Kubuntu 22.04
- GPU: 3080 16gb
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

The same commands using Franka reach instead of UR10 works fine. Also, with UR10, rsl-rl works fine.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] UR10 reach working with rl_games
- [ ] UR10 reach working with skrl

DJT777, 2024-06-21:
I have the same issue as well. Running on RSL RL, but not the others. Having RL Games would be beneficial for our work we are doing.

Mayankm96, 2024-06-25:
We do not provide configurations for all possible RL frameworks that we support. This is a tedious process as the number of environments keep growing and a lot of tuning efforts needs to be spent to get it work properly.

You can extend the support for RL-Games by:

1. Adding the configuration file for it inside the `agents` directory of that environment
2. Adding the entry point to that configuration while registering the environment

For example, for the lift environment:

1. The RL-Games configuration used there: https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/agents/rl_games_ppo_cfg.yaml
2. Specification of this file to the task registeration: https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/lift/config/franka/__init__.py#L25

DJT777, 2024-06-25:
I think the issue here is that there appears to be an existing config and environment for the UR10 for RL Games, but it's just not working for whatever reason.

https://github.com/isaac-sim/IsaacLab/tree/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/manipulation/reach/config/ur_10

eferreirafilho, 2024-06-25:
Exactly @DJT777 . I thought it was supposed to work as the config files are there.

DJT777, 2024-06-25:
@eferreirafilho 

I've been able to adapt the UR10 through basically modifying the Franka Each Env.

**source\extensions\omni.isaac.lab_tasks\omni\isaac\lab_tasks\manager_based\manipulation\reach\config\ur10-franka copy\joint_pos_env_cfg.py**

```
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

import math

from omni.isaac.lab.utils import configclass

import omni.isaac.lab_tasks.manager_based.manipulation.reach.mdp as mdp
from omni.isaac.lab_tasks.manager_based.manipulation.reach.reach_env_cfg import ReachEnvCfg

##
# Pre-defined configs
##
from omni.isaac.lab_assets import UR10_CFG_FRANKA  # isort: skip

##
# Environment configuration
##

@configclass
class UR10_FrankaReachEnvCfg(ReachEnvCfg):
    def __post_init__(self):
        # post init of parent
        super().__post_init__()

        # switch robot to franka
        self.scene.robot = UR10_CFG_FRANKA.replace(prim_path="{ENV_REGEX_NS}/Robot")
        # override events
        self.events.reset_robot_joints.params["position_range"] = (0.75, 1.25)
        # override rewards
        self.rewards.end_effector_position_tracking.params["asset_cfg"].body_names = ["ee_link"]
        self.rewards.end_effector_position_tracking_fine_grained.params["asset_cfg"].body_names = ["ee_link"]
        self.rewards.end_effector_orientation_tracking.params["asset_cfg"].body_names = ["ee_link"]
        # override actions
        self.actions.arm_action = mdp.JointPositionActionCfg(
            asset_name="robot", joint_names=[".*"], scale=0.5, use_default_offset=True
        )
        # override command generator body
        # end-effector is along x-direction
        self.commands.ee_pose.body_name = "ee_link"
        self.commands.ee_pose.ranges.pitch = (math.pi / 2, math.pi / 2)

@configclass
class UR10_FrankaReachEnvCfg_PLAY(UR10_FrankaReachEnvCfg):
    def __post_init__(self):
        # post init of parent
        super().__post_init__()
        # make a smaller scene for play
        self.scene.num_envs = 50
        self.scene.env_spacing = 2.5
        # disable randomization for play
        self.observations.policy.enable_corruption = False
```

source\extensions\omni.isaac.lab_tasks\omni\isaac\lab_tasks\manager_based\manipulation\reach\config\ur10-franka copy\agents\rl_games_ppo_cfg.yaml

```
params:
  seed: 42

  # environment wrapper clipping
[... 162 lines cut ...]
        joint_pos={
            "shoulder_pan_joint": 0.0,
            "shoulder_lift_joint": -1.712,
            "elbow_joint": 1.712,
            "wrist_1_joint": 0.0,
            "wrist_2_joint": 0.0,
            "wrist_3_joint": 0.0,
        },
        pos=(0.0, 0.0, 1), rot=(0,1,0,0)
    ),
    actuators={
        "arm": ImplicitActuatorCfg(
            joint_names_expr=[".*"],
            velocity_limit=100.0,
            effort_limit=87.0,
            stiffness=800.0,
            damping=40.0,
        ),

    },
    soft_joint_pos_limit_factor=1.0,
)

TELAMON_FRANKA_PANDA_HIGH_PD_CFG = UR10_CFG_FRANKA.copy()
```

DJT777, 2024-06-25:
@eferreirafilho 

It's a workaround, but the bigger issue is something within the UR10 Environments has a bug in it somewhere. This should get you going with the UR10 example though. Obviously not an ideal solution, since it is basically adapting the Franka environment. However, it's working.

Please don't mind my messy code!

eferreirafilho, 2024-06-25:
Thanks for sharing @DJT777 ! I will take a look

DJT777, 2024-06-25:
@eferreirafilho 

No problem, if this is a proper solution I'd like to make a proper PR to the repo so that the UR environment is working.

Also, you'll need to update the assets init.

source\extensions\omni.isaac.lab_assets\omni\isaac\lab_assets\__init__.py

```
from .ur10_franka import *
```

sudhirpratapyadav, 2024-07-13:
I have found the solution for this bug.

**TL;DR:** Create a `__init__.py` file inside the `ur_10.agents` folder and paste below code in it.
```
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

from . import rsl_rl_ppo_cfg  # noqa: F401, F403
```

**Why this problem occured**
The actual problem is there is no `__init__.py` file in the `agents` folder inside ur_10. Therefore when `mod_path = os.path.dirname(importlib.import_module(mod_name).__file__)` line is executed it gives error as `importlib.import_module(mod_name).__file__` is not able to find the module `omni.isaac.lab_tasks.manager_based.manipulation.reach.config.ur_10.agents`. 

If we add a `__init__.py` file inside the `agents` folder then this issue will be resolved.

closed 2024-07-17 by Dhoeller19