isaac-sim/IsaacLab #543: [Question] Necessary procedure to import URDF files

catachiii, 2024-06-20:
### Description

Dear IsaacLab team. Thanks for providing us with up-to-date USD files for all kinds of humanoids. I have couple of questions about the necessary procedures to import the URDFs by myself. 

Taking the URDFs of Unitree H1 for example (https://github.com/unitreerobotics/unitree_ros/tree/master/robots/h1_description/urdf), I have been trying to convert the given URDF to USD files with the How-to guide (https://isaac-sim.github.io/IsaacLab/source/how-to/import_new_asset.html). 

I successfully converted the URDF with `--merge-joints` and `--make-instanceable`. However, when I'm loading the humanoid USD file similarly to the demos (https://github.com/isaac-sim/IsaacLab/blob/main/source/standalone/demos/bipeds.py). I encountered the issue that some of the visuals are not updated with physics, as shown below. 

[image]

I'm not sure what I missed when converting. So far, all quadruped URDFs from Unitree work fine and similarly to what is provided from the Nucleus server. 

Thanks for your help!

Jin

fuyh20, 2024-06-20:
Hello, I've got the converted h1_with_hand USD file, see the file [h1_with_hand.zip](https://github.com/user-attachments/files/15910916/h1_with_hand.zip).

I had the same problem as you, when using the script, convert_urdf.py. But I used isaac-sim GUI, the menu bar Isaac Utils -> Workflows -> URDF Importer, and then got the correct USD file.

I hope this helps!

fuyh20, 2024-06-20:
hello, can you try to use this USD file?
[h1_with_hand.zip](https://github.com/user-attachments/files/15911814/h1_with_hand.zip)
I tested it, and it can work well. This way you can tell if there's a problem with the file or with the code.

catachiii, 2024-06-20:
@fuyh20 It looks good to me although I think it is wrongly converted with fixed base, so it is hard to say. 
[image]

catachiii, 2024-06-20:
Update:

I tried without `--merge_joints`, and now the visuals look all good.