isaac-sim/IsaacLab #546: Importing Carter from URDF does not allow actuation, no textures/colors

jbt-cmu, 2024-06-20:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

When using the Omniverse URDF importer to create a USD file, the Carter robot is all-white and when attempting to actuate the wheels it doesn't move. Something similar happens when attempting Hello-Robot's Stretch Bot

### Steps to reproduce

Here is my code: 

```
import argparse
from omni.isaac.lab.app import AppLauncher

# Add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on adding sensors on a robot.")
parser.add_argument("--num_envs", type=int, default=1, help="Number of environments to spawn.")
# Append AppLauncher CLI args
AppLauncher.add_app_launcher_args(parser)
# Parse the arguments
args_cli = parser.parse_args()

# Launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

import torch
import omni.usd
import numpy as np
import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import Articulation, ArticulationCfg, AssetBaseCfg, RigidObjectCfg
from omni.isaac.lab.scene import InteractiveScene, InteractiveSceneCfg
from omni.isaac.lab.utils import configclass
from omni.isaac.lab.envs import ManagerBasedEnv, ManagerBasedEnvCfg
from omni.isaac.lab.sim import CollisionPropertiesCfg
from omni.isaac.lab.actuators.actuator_cfg import ImplicitActuatorCfg
from omni.isaac.lab.managers import EventTermCfg as EventTerm
import omni.isaac.lab.envs.mdp as mdp

usd_path = "C:/Users/Owner/Desktop/stretch/carter/carter.usd"

# Define constant forward velocity
def constant_forward_velocity(env: ManagerBasedEnv) -> torch.Tensor:
    """Generates constant forward velocity command."""
    return torch.tensor([[10.0, 10.0]], device=env.device).repeat(env.num_envs, 1)

@configclass
class SceneCfg(InteractiveSceneCfg):
    """Design the scene with sensors on the robot."""
    # Ground plane
    ground = AssetBaseCfg(prim_path="/World/defaultGroundPlane", spawn=sim_utils.GroundPlaneCfg())

    # Lights
    dome_light = AssetBaseCfg(
        prim_path="/World/Light", spawn=sim_utils.DomeLightCfg(intensity=2000.0, color=(0.75, 0.75, 0.75))
    )

    # Robot
    robot= ArticulationCfg(
        prim_path="/World/Robot",
        spawn=sim_utils.UsdFileCfg(
            usd_path=usd_path,
            activate_contact_sensors=True  # Ensure contact sensors are activated
        ),
        init_state=ArticulationCfg.InitialStateCfg(pos=(0.0, 0.0, 4), rot=(0, 0, 0, 1)),
        actuators={
            "left_wheel": ImplicitActuatorCfg(joint_names_expr="left_wheel", stiffness=0.0, damping=0.0, effort_limit=1000),
            "right_wheel": ImplicitActuatorCfg(joint_names_expr="right_wheel", stiffness=0.0, damping=0.0, effort_limit=1000),
        },
    )

@configclass
class ObservationsCfg:
    pass

@configclass
class EventCfg:
    """Configuration for events."""
    reset_scene = EventTerm(func=mdp.reset_scene_to_default, mode="reset")

@configclass
class ActionsCfg:
    """Action specifications for the MDP."""
    wheel_velocity = mdp.JointVelocityActionCfg(
        asset_name="robot",
        joint_names=['left_wheel', 'right_wheel'],
        scale=100.0,  # Increased scale for higher actuation
    )

@configclass
class StretchEnvCfg(ManagerBasedEnvCfg):
    scene: SceneCfg = SceneCfg(num_envs=1, env_spacing=2.5)
    observations: ObservationsCfg = ObservationsCfg()
    actions: ActionsCfg = ActionsCfg()
    events: EventCfg = EventCfg()

    def __post_init__(self):
        self.sim.dt = 0.005
        self.decimation = 4

def main():

    env_cfg = StretchEnvCfg()
    env = ManagerBasedEnv(cfg=env_cfg)

    # Initialize the robot and actuators
    #env.scene.robot = Articulation(env_cfg.scene.robot)
    # env.scene.robot.initialize()

    # Print all joints and links
    print("All joint names:", env.scene['robot'].joint_names)
    print("All link names:", env.scene['robot'].body_names)

    env.reset()
    while simulation_app.is_running():
        # Print joint positions and velocities before the action
        print(f"Left wheel joint position: {env.scene['robot'].data.joint_pos[:, 0]}")
        print(f"Right wheel joint position: {env.scene['robot'].data.joint_pos[:, 1]}")
        print(f"Left wheel joint velocity: {env.scene['robot'].data.joint_vel[:, 0]}")
        print(f"Right wheel joint velocity: {env.scene['robot'].data.joint_vel[:, 1]}")

        # Apply constant forward velocity to the wheels
        action = constant_forward_velocity(env)
        env.step(action)

        # Print joint positions and velocities after the action
        print(f"After action:")
        print(f"Left wheel joint position: {env.scene['robot'].data.joint_pos[:, 0]}")
        print(f"Right wheel joint position: {env.scene['robot'].data.joint_pos[:, 1]}")
        print(f"Left wheel joint velocity: {env.scene['robot'].data.joint_vel[:, 0]}")
        print(f"Right wheel joint velocity: {env.scene['robot'].data.joint_vel[:, 1]}")
        print(f"Action applied to wheels: {action}")

        # Check joint limits and actuation efforts
        left_wheel_limits = env.scene['robot'].root_physx_view.get_dof_limits()
        right_wheel_limits = env.scene['robot'].root_physx_view.get_dof_limits()
        left_wheel_effort = env.scene['robot'].root_physx_view.get_dof_max_forces()
        right_wheel_effort = env.scene['robot'].root_physx_view.get_dof_max_forces()

        print(f"Joint Limits and Actuation Efforts:")
        print(f"Left wheel joint limits: {left_wheel_limits}")
        print(f"Right wheel joint limits: {right_wheel_limits}")
        print(f"Left wheel actuation effort: {left_wheel_effort}")
        print(f"Right wheel actuation effort: {right_wheel_effort}")

    env.close()

if __name__ == "__main__":
    main()
    simulation_app.close()

```
note you will need to retrieve the Carter USD and change the usd_path above. My error is that Carter is untextured and the wheels do not actuate. Does anything look wrong with my code?

[image]

### System Info

Describe the characteristic of your environment:

- Commit: 9ab6b485bad1b57fb93d119907d749d4d2bf289e
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: Windows 11
- GPU: RTX 4090
- CUDA: N/A
- GPU Driver: N/A

### Additional context

Add any other context about the problem here.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Carter successfully Actuates
- [ ] Importing from URDF potentially permits textures

Mayankm96, 2024-07-01:
I checked the Stretch robot URDF, and it seems to use STL files that don't specify textures. Hence, when you try to import the URDF, you won't see any textures. You will need to manually add textures to make it look nice.

For Carter, at least the URDF in the URDF importer example has the same issue. The OBJ meshes don't specify a color/texture on them.

For actuation, you need to set the damping of the drive for them to take into effect. I modified the script for that:

```python
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

import argparse

from omni.isaac.lab.app import AppLauncher

# Add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on adding sensors on a robot.")
parser.add_argument("--num_envs", type=int, default=1, help="Number of environments to spawn.")
# Append AppLauncher CLI args
AppLauncher.add_app_launcher_args(parser)
# Parse the arguments
args_cli = parser.parse_args()

# Launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

import os
import torch

from omni.isaac.core.utils.extensions import enable_extension, get_extension_path_from_name

import omni.isaac.lab.envs.mdp as mdp
import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.actuators.actuator_cfg import ImplicitActuatorCfg
from omni.isaac.lab.assets import ArticulationCfg, AssetBaseCfg
from omni.isaac.lab.envs import ManagerBasedEnv, ManagerBasedEnvCfg
from omni.isaac.lab.managers import EventTermCfg as EventTerm
from omni.isaac.lab.scene import InteractiveSceneCfg
from omni.isaac.lab.utils import configclass

# Define constant forward velocity
def constant_forward_velocity(env: ManagerBasedEnv) -> torch.Tensor:
    """Generates constant forward velocity command."""
    return torch.tensor([[10.0, 10.0]], device=env.device).repeat(env.num_envs, 1)

@configclass
class SceneCfg(InteractiveSceneCfg):
    """Design the scene with sensors on the robot."""

    # Ground plane
    ground = AssetBaseCfg(prim_path="/World/defaultGroundPlane", spawn=sim_utils.GroundPlaneCfg())

    # Lights
    dome_light = AssetBaseCfg(
        prim_path="/World/Light", spawn=sim_utils.DomeLightCfg(intensity=2000.0, color=(0.75, 0.75, 0.75))
    )

    # Robot
    robot = ArticulationCfg(
        prim_path="/World/Robot",
        spawn=sim_utils.UrdfFileCfg(
            asset_path="carter.urdf",  # filled in post_init with full path from extension
            merge_fixed_joints=True,
            fix_base=False,
            articulation_props=sim_utils.ArticulationRootPropertiesCfg(enabled_self_collisions=False),
        ),
        init_state=ArticulationCfg.InitialStateCfg(pos=(0.0, 0.0, 0.25), rot=(1, 0, 0, 0)),
        actuators={
[... 72 lines cut ...]
        # print(f"Right wheel joint position: {env.scene['robot'].data.joint_pos[:, 1]}")
        # print(f"Left wheel joint velocity: {env.scene['robot'].data.joint_vel[:, 0]}")
        # print(f"Right wheel joint velocity: {env.scene['robot'].data.joint_vel[:, 1]}")
        # print(f"Action applied to wheels: {action}")

        # Check joint limits and actuation efforts
        # left_wheel_limits = env.scene['robot'].root_physx_view.get_dof_limits()
        # right_wheel_limits = env.scene['robot'].root_physx_view.get_dof_limits()
        # left_wheel_effort = env.scene['robot'].root_physx_view.get_dof_max_forces()
        # right_wheel_effort = env.scene['robot'].root_physx_view.get_dof_max_forces()

        # print(f"Joint Limits and Actuation Efforts:")
        # print(f"Left wheel joint limits: {left_wheel_limits}")
        # print(f"Right wheel joint limits: {right_wheel_limits}")
        # print(f"Left wheel actuation effort: {left_wheel_effort}")
        # print(f"Right wheel actuation effort: {right_wheel_effort}")

    env.close()

if __name__ == "__main__":
    main()
    simulation_app.close()
```

However, it should be noted that even with this fix, the wheel simulation is not great and the robot doesn't follow a straight path. Other users who are interested in wheel simulation have reported this. The PhysX team is looking into fixing this issue.

AntoineRichard, 2026-08-27:
Thanks for reporting this, @jbt-cmu — URDF import quirks are worth flagging.

A complete working script was posted in-thread that imports Carter through `sim_utils.UrdfFileCfg` and drives the wheels; the config in the original report had both `stiffness=0.0` and `damping=0.0`, so the wheels had nothing to hold them. The all-white look comes from the Carter/Stretch URDFs referencing OBJ/STL meshes that carry no material. Note the importer extension has since been renamed to `isaacsim.asset.importer.urdf`, so follow the current URDF import docs.

Closing as answered — reopen with a repro if the import itself still fails on a recent release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard