isaac-sim/IsaacLab #550: Mandatory to create Task for RL training

GRINGOLOCO7, 2024-06-22:
I was wondering, seeing the examples in the GitHub, all Reinforcment Learning algorithms has the structure of:
`./isaaclab.sh -p source/standalone/workflows/sb3/train.py --task Isaac-Cartpole-v0 --headless --cpu`

Without having the [template](https://isaac-sim.github.io/IsaacLab/source/setup/template.html) avaible at this [GitHub ](https://github.com/isaac-sim/IsaacLab.ext_template) that is inexisting, 

I WAS WONDERING, IF I NEED TO CREATE A TASK IN ORDER TO MAKE A REINFORCMENT LEARNING ALGORITM.
If yes, do I do this form [here](https://isaac-sim.github.io/IsaacLab/source/api/lab_tasks/omni.isaac.lab_tasks.utils.html)?

Mayankm96, 2024-06-23:
No, if you only care about the algorithm, then you need to implement that through your preferred choice of the library. You can then run the algorithm with the bash command similar to the one above.

If you want to integrate your own library, please check here: https://isaac-sim.github.io/IsaacLab/source/how-to/add_own_library.html

closed 2024-06-23 by Mayankm96

GRINGOLOCO7, 2024-06-24:
what if I want to train my own robot... This mean I won't use one of the avaible tasks (can be found with` ./isaaclab.sh -p source/standalone/environments/list_envs.py`).