isaac-sim/IsaacLab #553: Raycast tutorial: sensor height expected seems wrong?

agillies8, 2024-06-23:
In the Isaac Lab Tutorial 4 - Adding a sensor, in the raycaster example, Im trying to make sense of the measurement it returns, and it seems incorrect. After the ball is resting on the ground, I would expect the raycaster to return something close to the radius of the ball, which is 0.125. However, Im actualyl seeing it even out around 1.22, which I dont understand how thats possible.

[Screencast from 06-22-2024 10:56:20 PM.webm](https://github.com/isaac-sim/IsaacLab/assets/11578915/992936d7-6706-44c9-8c53-8dfd9294af72)

To simplify the setup, I reduced the number of balls to one, and fixed it so it always falls from a height of 10m.

### Steps to reproduce

I changed line 72 to:

origins = [[0.0, 0.0, 0.0]]

And added these lines at line 104:

    ball_default_state[:, 2] = 10
    ball_default_state[:,0] = 1
    ball_default_state[:,1] = 0.5

Then ran the sim, the max height the sensor reads is stil. 1.22. Why is this?

Mayankm96, 2024-06-23:
As mentioned [in the documentation](https://isaac-sim.github.io/IsaacLab/source/api/lab/omni.isaac.lab.sensors.html#omni.isaac.lab.sensors.RayCaster):

> Currently, only static meshes are supported. Extending the warp mesh to support dynamic meshes is a work in progress.

The raycaster in the above example only ray-casts against the terrain. It can be extended for dynamic cases (which includes the ball), but this is yet to be implemented. We don't have any immediate plans to add support since we are focusing more on improving the tiled rendering.

Zhefan-Xu, 2024-07-15:
@Mayankm96 Thanks for your insights. Is it possible to implement raycast for dynamic objects in Isaac Sim. I would appreciate it if you can provide some suggestions. Thank you!

closed 2025-01-05 by RandomOakForest