isaac-sim/IsaacLab #558: IsaacLab.ext_template

fighting-j, 2024-06-24:
When I build my own project according to the Isaaclab documentation, I open the extension module but get an error that the page cannot be found.
I get the following error when using git clone：
fatal: 'https://github.com/isaac-sim/IsaacLab.ext_template/' Authentication failed
When will the expansion module be released?

KyleM73, 2024-06-24:
the extension is here:

https://github.com/isaac-orbit/orbit.ext_template

it requires a little work to make it compatible with current isaac lab, changing the names of the packages in the toml/setup files, etc. it will likely be updated but hasn't yet.

jsmith-bdai, 2024-06-24:
The Isaac Lab team is actively working on updating this Extension Template, so it will be available very soon! Sorry for any convenience in the meantime

Closing as this is a duplicate of https://github.com/isaac-sim/IsaacLab/issues/457

closed 2024-06-24 by jsmith-bdai